brocaar/chirpstack-application-server #1: Network Operations Centre with key-metri

vannut, 2016-09-27:


hecko, 2016-11-20:
@vannut the infromation are available by subscribing to the gateway MQTT topic - we use this stream of data to push information to our own monitoring system + use zabbix to monitor the gateways. This is a very neat way to integrate the loraserver to your NOC solution. (easy to do for zabbix, nagios...)

vannut, 2016-11-21:
Never new there was a Gateway MQtt topic :) 
What would be the default data in such a message? Would the data sent in that topic be customizable? 
I do not have a working instance of the lora-app-server running; so i'm not able to check it.

hecko, 2016-11-22:
The data sent to that topis are almost identical (if not completely identical) to the data that packet forwarder sends out via UDP port.

brocaar, 2017-03-23:
Guys, I'm looking forward to your feedback on https://github.com/brocaar/lora-app-server/pull/42! I've added some screenshots to give you an idea of the upcoming changes.

brocaar, 2017-03-28:
Gateway management + stats has been released!

closed 2017-04-25 by brocaar