brocaar/chirpstack-application-server #4: JWT, I need instructions or it is broken?

siscia, 2016-10-16:
Hi,

I am having a lot of problem with JWT, I either need instruction or it is broken.

Let's move with order.

1) To generate the JWT I use jwt-go, doing something like this:
```
$ cat key
secret
$ cat api.json 
{    "admin": true,                
    "apis": ["*"],          
    "apps": ["*"],  
    "nodes": ["*"]                 
}
$ jwt -key key -alg HS256 -sign api.json
eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhZG1pbiI6dHJ1ZSwiYXBpcyI6WyIqIl0sImFwcHMiOlsiKiJdLCJub2RlcyI6WyIqIl19.46ppnIA4qJ4GgEC5awYMU3Km3qekVNUv6dGY4lhe6Pk
$ jwt -key key -alg HS256 -sign api.json | jwt -show -
Header:
{
    "alg": "HS256",
    "typ": "JWT"
}
Claims:
{
    "admin": true,
    "apis": [
        "*"
    ],
    "apps": [
        "*"
    ],
    "nodes": [
        "*"
    ]
}
$ jwt -key key -alg HS256 -sign api.json | jwt -key key -verify -
{
    "admin": true,
    "apis": [
        "*"
    ],
    "apps": [
        "*"
    ],
    "nodes": [
        "*"
    ]
}
$ jwt -key key -alg HS256 -sign api.json | jwt -debug -key key -verify -
Token len: 153 bytes
Header:
map[alg:HS256 typ:JWT]
Claims:
map[admin:true apis:[*] apps:[*] nodes:[*]]
{
    "admin": true,
    "apis": [
        "*"
    ],
    "apps": [
        "*"
    ],
    "nodes": [
        "*"
    ]
}
```
Everything seems correct, however JWT.io don't authorize the token.
Not sure where is the problem but I have reported it to jwt-go: https://github.com/dgrijalva/jwt-go/issues/154#issuecomment-253979102

Unfortunately lora-app-server doesn't recognize the token neither.

On top of that I am not able to provide an expiration time for the token, it fails to parse, it is recognized as a float and is not parsed into the int64.

Now, it is just me doing something wrong, or JWT as really some problem?

brocaar, 2016-11-13:
Sorry for my late reply, I was on vacation. Are you sure the content of your `key` file does not contain a trailing `\n`? (you can use `cat -e key`, there must not be a trailing `$`).

When generating the token, I get `eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJhZG1pbiI6dHJ1ZSwiYXBpcyI6WyIqIl0sImFwcHMiOlsiKiJdLCJub2RlcyI6WyIqIl19.vE4rNylxprgNWNFnAdhXg5AWy5_9F4WfXxOsYRJjg5o` which is valid according to https://jwt.io.

brocaar, 2016-11-25:
@siscia did you manage to get it working?

siscia, 2016-11-29:
I haven't yet had the time to check it... Sorry for the delay.

siscia, 2016-12-07:
Works perfectly, no idea what I was doing wrong.

closed 2016-12-07 by siscia