brocaar/chirpstack-application-server #5: Node list not returned

kuripiiritus, 2016-10-24:
Hei

There seems to be some issue with retuning the node list. 
current config: loraserver on 0.12.3 lora-app-server on 0.1.3

When requesting the node list (GET /api/node) the result be as per description 
`{
  "result": [
    {
      "appEUI": "string",
      "appKey": "string",
      "channelListID": "string",
      "devEUI": "string",
      "name": "string",
      "relaxFCnt": true,
      "rx1DROffset": 0,
      "rx2DR": 0,
      "rxDelay": 0,
      "rxWindow": "RX1"
    }
  ],
  "totalCount": "string"
}`

unfortunately it only replies with 
`{
  "totalCount": "7"
}`
It gets the count, but does not display the actual nodes.

adding the offset and limit  (GET /api/node/0/100) returns not found. Doing the same in the 0.1.1 version server, works flawlessly.

Br,
Kuldar Loime

sopach, 2016-11-07:
Hi,

I Have exactly the same problem, with the same configuration.

Br.

Stanislav

brocaar, 2016-11-13:
Sorry for my late reply, I was on vacation. Please see the api documentation: 

[image]

You should use `?limit=10&offset=0` now :-) This makes the api a bit cleaner.

kuripiiritus, 2016-11-14:
Thanks for the reply. This does look better :)

hecko, 2016-12-27:
IMO the docs to this are still rather unclear - I had to search here to find the answer, which sort of demonstrates this solution (or rather information) should really be specifically mentioned in the API call description.

brocaar, 2017-01-29:
This will be properly fixed in the next version as url query arguments are now correctly documented in the Swagger specs (by https://github.com/grpc-ecosystem/grpc-gateway). 

[image]

commit 1bedb216 (2017-01-29): Update vendored packages / regenerate swagger specs.

Closes #5.
closed 2017-01-29 by brocaar