brocaar/chirpstack-application-server #6: Possible issue with frame count reaching 65535

andrepferreira, 2016-10-27:
Hi there

I have been running a test mote for a few days now and today I started getting the following error with rxInfo:

> 
{"devEUI":"0000000079250001","type":"DATA_UP_MIC","error":"invalid MIC"}
> 

The log entries are as follows:

> Oct 27 10:28:52 iotlab02 lora-app-server[15272]: time="2016-10-27T10:28:52+02:00" level=error msg="invalid MIC" dev_eui=0000000079250001 type=DATA_UP_MIC
Oct 27 10:28:52 iotlab02 lora-app-server[15272]: time="2016-10-27T10:28:52+02:00" level=info msg="handler/mqtt: publishing error notification" topic="application/0000000000000000/node/0000000079250001/error"
Oct 27 10:28:52 iotlab02 loraserver[15233]: time="2016-10-27T10:28:52+02:00" level=error msg="processing rx packet error: invalid MIC" data_base64="QAEAJXmAdg4BvV+qY82d0bcu7h9xMFhIAqm1iO3qCremu1DlZeztNWHaDsvqUCWZKg=="

> 

The only thing I noticed is that on the Lora-app-server, the frame counter seemed to have stopped at 65535.

Restarting the mote reset the frame counter and all transmissions resumed as normal.

Could the frame counter be overflowing, thereby causing the error. 

I'm using version **0.12.3** for the **loraserver** and **0.1.3** for the **apps** server.

andrepferreira, 2016-11-14:
Quick update. I have been able to replicate this issue as my mote has once again reached 65535 packets and the same MIC error above has appeared. Below is the view of the frame count from the App-Server node config page. 

(keys removed for security) 

[image]

brocaar, 2016-11-14:
@andrepferreira I'm just back from vacation so I haven't had the change to look into it sooner. Anyway, I'm able to replicate it and will look into a fix asap.

brocaar, 2016-11-14:
I've found the issue. I might have introduced this bug during splitting the lora app server logic out of loraserver. What happens is:

The counter (32 bit) is incremented to 65536 at the side of the node (this value is used for the MIC) and because of this, the `FCnt` which is 16 bit is transmitted as 0, as it contains the 16 LSB. The server is able to detect the 16 bit rollover, but because the MIC validation is happening before the `FCnt` expansion, it is using the wrong `FCnt` to validate the MIC (so it is validated against 0 instead of 65536).

I have a fix working locally and will add a testcase to make sure this regression won't be re-introduced again :-)

brocaar, 2016-11-14:
See https://github.com/brocaar/loraserver/commit/2849d0298ce44153876b878183043b5a5f5b9747 for the fix. Once the build is green, it will be released as loraserver 0.12.4. Thanks for reporting this bug!

closed 2016-11-14 by brocaar

nickumbe, 2017-12-06:
Any chance this bug has been reintroduced?  I'm seeing the same issue on the latest version of Lora Server.