brocaar/chirpstack-application-server #9: Error grpc: the connection is unavailable (code: 14)

hecko, 2016-11-18:
Im getting this error when clicking on 'generate' in 
Dashboard / Nodes / xx0803xxxfd5b370 / node-session

Also, Im seeing a lot of, what seems like, bogus information in the last checkout....

Like for example - what do these mean? ->

loraserver       | time="2016-11-18T22:59:16Z" level=error msg="send rx info to network-controller error: publish rxinfo to network-controller error: rpc error: code = 12 desc = unknown service nc.NetworkController" dev_eui=000000008a68a920

and this is just strange:

loraserver       | time="2016-11-18T23:02:50Z" level=info msg="backend/gateway: rx packet received"
loraserver       | time="2016-11-18T23:02:50Z" level=info msg="packet(s) collected" dev_eui=490803f08fd5b370 gw_count=1 gw_macs=b827ebfffed1fcc2 mtype=JoinRequest
loraappserver    | time="2016-11-18T23:02:50Z" level=error msg="join-request node does not exist" dev_eui=490803f08fd5b370
loraserver       | time="2016-11-18T23:02:50Z" level=error msg="processing rx packet error: application server join-request error: rpc error: code = 2 desc = get node 490803f08fd5b370 error: sql: no rows in result set" data_base64="AKq7zN3u/xEicLPVj/ADCEns2So2+UU="

- the "does not exist" bit is for example wrong as the DevEUI actually exists in the database...

Im trying to find my way around this but without some more information this will be too much effort spent for finding my implementation issues. 

Thanks for all the help you can offer!

hecko, 2016-11-18:
OK, i think part of this issue is that information displayed in the web interface does not correlate with information I have in the postgres database. The postgres database has only one entry out of three in the web interface. How is this done that information is neither in redis, not in postgres but it still shows in the web interface? I feel like a total n00b here.

closed 2016-11-18 by hecko

RogerioCassares, 2017-08-10:
Hi @hecko , i'm having the same problem. could you help me, please? I'm in a big trouble :/

What you have done to soluntion this case, please?

brocaar, 2017-08-15:
@RogerioCassares https://forum.loraserver.io/ is a better place for support questions 👍