brocaar/chirpstack-application-server #10: Letsencrypt SSL certificates

drhexium, 2016-11-21:
Using the letsencrypt service you can obtain the following files:
cert.pem  chain.pem  fullchain.pem  privkey.pem

Can you please let me know which of these (and how) should be mapped onto the following environment variables?
HTTP_TLS_CERT
HTTP_TLS_KEY

Also, do any of the files .pem above need to be associated with the LoRaWAN server (or all associations need to be performed on the LoRa-app server)?

Thanks

hecko, 2016-11-22:
Im not entirely sure, but you can try to use the fullchain.pem for CERT and privkey.pem for KEY

Marcel

brocaar, 2016-11-22:
@hecko yes that is correct.

@drhexium `HTTP_TLS_CERT` and `HTTP_TLS_KEY` should contain the paths to these files.

drhexium, 2016-11-22:
It worked (I was trying to use cert.pem on HTTP_TLS_CERT previously)! 

This may be worth adding to the "Getting Started" documentation section.

Many thanks!

brocaar, 2016-11-25:
Great!

closed 2016-11-25 by brocaar