brocaar/chirpstack-application-server #11: Payload receive wrong

sabas1080, 2016-11-22:
Hi

I have a problem, i am sending the payload "AAAAAA" and I get this

```
{"devEUI":"0012000000000001","time":"2016-11-22T21:39:24.860463Z","fPort":50,"gwCount":1,"rssi":-52,"data":"qqqqqg=="}
qos : 0, retain : false, cmd : publish, dup : false, topic : application/70b3d57ed0000dd4/node/0012000000000001/rx, messageId : , length : 173, Raw payload : 123341001011186985733458344848495048484848484848484848484934443411610510910134583450484954454949455050845049585157585052465654485254519034443410280111114116345853484434103119671111171101163458494434114115115105345845535044341009711697345834113113113113113103616134125
```

¿"qqqqqg==" not is equal to AAAAAA in base64?

Thanks

brocaar, 2016-11-23:
I assume AAAAAA is HEX encoded. To compare you need to decode that back to
bytes en re-encode these bytes to base64.

hecko, 2016-11-23:
qqqqqg== is equal to AAAAAAAA in hex (also known as 0xAAAAAAAA or 0xAA 0xAA 0xAA 0xAA)

Online base64 to HEX converter: http://tomeko.net/online_tools/base64.php?lang=en

Marcel

sabas1080, 2016-11-23:
Thank you very much @brocaar @hecko

closed 2016-11-23 by sabas1080

grendy, 2017-07-17:
Hi,
Can you explain how to read data from lora-app-server gui?
I subscribe from application/+/node/+/rx mqtt broker, and i got json 
{
  "applicationID" : "1",
  "applicationName" : "Lora",
  "nodeName" : "Lora",
  "devEUI" : "33393437673xxxx",
  "rxInfo" : [ {
    "mac" : "7276ff000b0xxxxx",
    "rssi" : -62,
    "loRaSNR" : 11,
    "name" : "Gateway",
    "latitude" : -x.xxxxx,
    "longitude" : xxx.xxxxx,
    "altitude" : 40
  } ],
  "txInfo" : {
    "frequency" : 923200000,
    "dataRate" : {
      "modulation" : "LORA",
      "bandwidth" : 125,
      "spreadFactor" : 10
    },
    "adr" : true,
    "codeRate" : "4/5"
  },
  "fCnt" : 726,
  "fPort" : 2,
  "data" : "CPw="
}
GUI lora-app-server:
[image]

Thanks before.