brocaar/chirpstack-application-server #13: TX data queue never clears?

jcwren, 2016-11-29:
This weekend I brought up the lora-gateway-bridge, lora-app-server, and loraserver on a Raspberry Pi 3, with a Haxiot Micro LoRaWAN Gateway. I'm using RN2903's for clients. Everything is playing nicely (including OTAA!), with one problem I haven't been able to resolve.

When I queue up a packet for transmission via MQTT, it gets sent to the client, but it gets sent every time the radio transmits. As far as I can tell, the data should only be sent once. 

The only deviation in my installation is that I'm running the MQTT server on another machine. I'm able to receive and transmit packets from that server, so I don't think it's anything related to that.

Do I need to enable something in the radio to acknowledge it? I'm happy to provide any additional debugging output, if someone can indicate what they'd like to see.

brocaar, 2016-11-29:
Could you add some more data. E.g. did you set `confirmed: true` for the downlink payload? Also it would help to post the log outputs of all the components. After a transmission (and in case of confirmed=true), the payload should be cleared from the queue.

jcwren, 2016-11-30:
I've tried this with and without enabling automatic replies ('mac set ar on', 'mac set ar off'), and no difference.

```
$ mosquitto_sub -v -t 'gateway/#' -t "application/#"
application/0807060504030201/node/0004a30b001a2d83/tx {"reference":"abcd1234", "confirmed": true, "devEUI": "0004a30b001a2d83", "fport": 1, "data": "QUJE"}
gateway/080027ffff64cc52/rx {"rxInfo":{"mac":"080027ffff64cc52","time":"2016-11-30T02:15:03.125633Z","timestamp":514715835,"frequency":902300000,"channel":0,"rfChain":0,"crcStatus":1,"codeRate":"4/5","rssi":101,"loRaSNR":10,"size":16,"dataRate":{"modulation":"LORA","spreadFactor":7,"bandwidth":125}},"phyPayload":"gGr41QYgAQAB0FWuN6k+4A=="}
application/0807060504030201/node/0004a30b001a2d83/rx {"devEUI":"0004a30b001a2d83","rxInfo":[{"mac":"080027ffff64cc52","time":"2016-11-30T02:15:03.125633Z","rssi":101,"loRaSNR":10}],"fCnt":1,"fPort":1,"data":"QUJD"}
application/0807060504030201/node/0004a30b001a2d83/ack {"reference":"abcd1234","devEUI":"0004a30b001a2d83"}
gateway/080027ffff64cc52/tx {"txInfo":{"mac":"080027ffff64cc52","immediately":false,"timestamp":515715835,"frequency":923300000,"power":20,"dataRate":{"modulation":"LORA","spreadFactor":7,"bandwidth":500},"codeRate":"4/5","iPol":null},"phyPayload":"oGr41QYwAQABOs7hflDu0Q=="}
```

```
Nov 30 02:14:47 rpi3 lora-app-server[1053]: time="2016-11-30T02:14:47Z" level=info msg="handler/mqtt: data-down payload received" topic="application/0807060504030201/node/0004a30b001a2d83/tx"
Nov 30 02:14:47 rpi3 lora-app-server[1053]: time="2016-11-30T02:14:47Z" level=info msg="downlink queue item enqueued" dev_eui=0004a30b001a2d83 id=4
Nov 30 02:14:50 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:14:50Z" level=info msg="gateway: received udp packet from gateway" addr=127.0.0.1:56268 protocol_version=1 type=PullData
Nov 30 02:14:50 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:14:50Z" level=info msg="gateway: sending udp packet to gateway" addr=127.0.0.1:56268 protocol_version=1 type=PullACK
Nov 30 02:15:00 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:15:00Z" level=info msg="gateway: received udp packet from gateway" addr=127.0.0.1:56268 protocol_version=1 type=PullData
Nov 30 02:15:00 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:15:00Z" level=info msg="gateway: sending udp packet to gateway" addr=127.0.0.1:56268 protocol_version=1 type=PullACK
Nov 30 02:15:03 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:15:03Z" level=info msg="gateway: received udp packet from gateway" addr=127.0.0.1:39018 protocol_version=1 type=PushData
Nov 30 02:15:03 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:15:03Z" level=info msg="gateway: rxpk packet received" addr=127.0.0.1:39018 data="gGr41QYgAQAB0FWuN6k+4A==" mac=080027ffff64cc52
Nov 30 02:15:03 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:15:03Z" level=info msg="backend: publishing packet" topic="gateway/080027ffff64cc52/rx"
Nov 30 02:15:03 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:15:03Z" level=info msg="gateway: sending udp packet to gateway" addr=127.0.0.1:39018 protocol_version=1 type=PushACK
Nov 30 02:15:03 rpi3 loraserver[1068]: time="2016-11-30T02:15:03Z" level=info msg="backend/gateway: rx packet received"
Nov 30 02:15:03 rpi3 loraserver[1068]: time="2016-11-30T02:15:03Z" level=info msg="packet(s) collected" dev_eui=0004a30b001a2d83 gw_count=1 gw_macs=080027ffff64cc52 mtype=ConfirmedDataUp
Nov 30 02:15:03 rpi3 loraserver[1068]: time="2016-11-30T02:15:03Z" level=info msg="rx info sent to network-controller" dev_eui=0004a30b001a2d83
Nov 30 02:15:03 rpi3 lora-app-server[1053]: time="2016-11-30T02:15:03Z" level=info msg="handler/mqtt: publishing data-up payload" topic="application/0807060504030201/node/0004a30b001a2d83/rx"
Nov 30 02:15:03 rpi3 loraserver[1068]: time="2016-11-30T02:15:03Z" level=info msg="node-session saved" dev_addr=06d5f86a
Nov 30 02:15:03 rpi3 lora-app-server[1053]: time="2016-11-30T02:15:03Z" level=info msg="downlink queue item deleted" id=1
Nov 30 02:15:03 rpi3 lora-app-server[1053]: time="2016-11-30T02:15:03Z" level=info msg="downlink queue item acknowledged" dev_eui=0004a30b001a2d83
Nov 30 02:15:03 rpi3 lora-app-server[1053]: time="2016-11-30T02:15:03Z" level=info msg="handler/mqtt: publishing ack notification" topic="application/0807060504030201/node/0004a30b001a2d83/ack"
Nov 30 02:15:03 rpi3 loraserver[1068]: time="2016-11-30T02:15:03Z" level=info msg="node-session saved" dev_addr=06d5f86a
Nov 30 02:15:03 rpi3 lora-app-server[1053]: time="2016-11-30T02:15:03Z" level=info msg="downlink queue item updated" id=2
Nov 30 02:15:03 rpi3 lora-app-server[1053]: time="2016-11-30T02:15:03Z" level=info msg="data-down item requested by network-server" confirmed=true dev_eui=0004a30b001a2d83 fcnt=1 id=2
Nov 30 02:15:03 rpi3 loraserver[1068]: time="2016-11-30T02:15:03Z" level=info msg="received data down from application" confirmed=true data_base64=Os7h dev_eui=0004a30b001a2d83 fcnt=1 more_data=true
Nov 30 02:15:03 rpi3 loraserver[1068]: time="2016-11-30T02:15:03Z" level=info msg="backend/gateway: publishing tx packet" topic="gateway/080027ffff64cc52/tx"
Nov 30 02:15:03 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:15:03Z" level=info msg="backend: packet received" topic="gateway/080027ffff64cc52/tx"
Nov 30 02:15:03 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:15:03Z" level=info msg="gateway: sending udp packet to gateway" addr=127.0.0.1:56268 protocol_version=1 type=PullResp
Nov 30 02:15:11 rpi3 lora-gateway-bridge[1042]: time="2016-11-30T02:15:11Z" level=info msg="gateway: received udp packet from gateway" addr=127.0.0.1:56268 protocol_version=1 type=PullData
```

brocaar, 2016-11-30:
Could you also check how many items are in the queue? For this you can
check directly in the PostgreSQL db (downlink_queue table). From the above
logs it seems the ACK is handled correctly but there might be multiple /
duplicated items in the queue.

jcwren, 2016-12-01:
It appears we can close this. I'm not sure whether I might have sent the TX data more times than I thought, or after restarting the entire system the problem cleared up (magic!). But at any rate, the message did survive the restart (that's good), and after the module sent a packet and received the queued message, the message was then cleared out of the table (even better).

I would have liked to have checked the counts in the table before I restarted it, but the Haxiot gateway had failed (USB connector snapped off the PCB), and I had to take down the RPi.

closed 2016-12-01 by jcwren