brocaar/chirpstack-application-server #16: Delete the content of the downlinkQueue

developy2, 2017-01-02:
Hello brocaar,
i want to syncronize the realtime clock from my nodes with a generated downlink packet.
Now i have the problem that sometimes more than one packet is stored in the downlinkQueue.
The result is an old timestamp (the oldest packet in the queue , FIFO) and a wrong clocktime at my nodes.
So, i want to clear the downlinkQueue. I write a curl command direktly at the raspberry pi like this:
curl -X DELETE --insecure --header 'Accept: application/json' 'https://xxx.xx.xx.xxx:8080/api/downlinkQueue/0004a30b001a2c98'
The answer is:
{"Error":"strconv.ParseInt: parsing \"0004a30b001a2c98\": invalid syntax","Code":2}
after that the queue is not empty. 
What is my mistake. Is there a better way to clean the queue?
It is possible to define only one entry size for tghe queue?
Thank you very much for help and for your nice work.

brocaar, 2017-01-21:
Hi @developy2, sorry for my late response I overlooked your issue. This might be a misunderstanding of the queue API. Instead of the `devEUI`, the id of the item in the queue is expected.

Maybe the API can be improved by:

- Returning the queue item id after a create
- `DELETE /api/downlinkQueue/0102030405060708` = delete the queue
- `DELETE /api/downlinkQueue/0102030405060708/123` delete queue item 123

What do you think?

Richaagarwalrn, 2017-05-20:
it should be like this ...Their should be one command to delete all queues related to particular device and selective queue item of that particular device.

Yaoshicn, 2017-07-17:
Hi, I am wondering if there is any way to delete all message in downlink queue?

brocaar, 2017-07-17:
Not yet.

Yaoshicn, 2017-07-17:
Thank you for your reply. I may writer a bash script to help.

brocaar, 2017-12-07:
I've just released LoRa App Server 0.15.0 + LoRa Server 0.23.0. This release includes a refactor of the downlink queue and also implements an API method to flush the entire downlink queue :-)

closed 2017-12-07 by brocaar