brocaar/chirpstack-application-server #18: tls: first record does not look like a TLS handshake

pablocaviglia, 2017-01-23:
When i try to connect to client api server through default port 8080 i got this.

**lora-app-server_1      | 2017/01/23 13:23:56 http: TLS handshake error from 172.19.0.1:55770: tls: first record does not look like a TLS handshake
lora-app-server_1      | 2017/01/23 13:24:07 http: TLS handshake error from 172.19.0.1:55772: EOF**

I have those environments variables configured on lora-app-server docker instance.
      - HTTP_TLS_CERT=/app/cert/lora-app-server.pem
      - HTTP_TLS_KEY=/app/cert/lora-app-server.pem

The docker initialization log is this:

**lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="starting LoRa App Server" docs="https://docs.loraserver.io/" version=0.2.0 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="connecting to postgresql" 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="setup redis connection pool" 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="handler/mqtt: connecting to mqtt broker" server="tcp://mosquitto:1883" 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="connecting to network-server api" ca-cert= server="loraserver:8000" tls-cert= tls-key= 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="applying database migrations" 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="handler/mqtt: connected to mqtt broker" 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="handler/mqtt: subscribling to tx topic" topic="application/+/node/+/tx" 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="migrations applied" count=0 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="starting application-server api" bind="0.0.0.0:8001" ca-cert= tls-cert= tls-key= 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=warning msg="client api authentication and authorization is disabled (set jwt-secret to enable)" 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="starting client api server" bind="0.0.0.0:8080" tls-cert="/app/cert/lora-app-server.pem" tls-key="/app/cert/lora-app-server.pem" 
lora-app-server_1      | time="2017-01-23T13:23:51Z" level=info msg="registering rest api handler and documentation endpoint" path="/api"** 

I've imported the corresponding p12 to the browser i'm using to connect to the api server.

brocaar, 2017-01-23:
Make sure use http_**s**_:// when making api requests or opening the web-interface

pablocaviglia, 2017-01-23:
Excellent man, very simple solution.

:+1:

closed 2017-01-23 by pablocaviglia

l0ng09, 2017-09-15:
So good man

dfree951, 2020-02-19:
@brocaar using http_s_://... and not getting good results