brocaar/chirpstack-application-server #22: Make HTTPS optional

drasko, 2017-02-11:
Hello,
currently TLS is mandatory: https://github.com/brocaar/lora-app-server/blob/master/cmd/lora-app-server/main.go#L121-L131 and this demands providing TLS cert into the production container.

When deployed on the trusted network TLS is not mandatory and should be configured as optional.

brocaar, 2017-02-11:
The reason behind is that in order to to enable http/2 support in the Go http server, TLS is required. In order to support the gRPC API (which uses http/2) and REST api running on the same port, this means that TLS must be enabled. See also: https://golang.org/pkg/net/http/

drasko, 2017-02-11:
Ah, thanks.

OK, I've added certs to proceed with Mainflux integration here: https://github.com/mainflux/mainflux/tree/master/lora

All services seem to be up and running now...

closed 2017-02-19 by brocaar