brocaar/chirpstack-application-server #23: How to config DB table?

mybrand, 2017-02-20:
[image]

When I try to access with web browser, after setup lora-app-server I got code:13 error message.
If I want to access app-server page, how to configure DB setting?

brocaar, 2017-02-21:
Please see the `--db-automigrate` config flag: https://docs.loraserver.io/lora-app-server/configuration/. When this flag is given, LoRa App Server will upgrade your database schema on start (see also: https://github.com/brocaar/lora-app-server/tree/master/migrations). So this not only creates all tables, but will perform schema updates when upgrading LoRa App Server.

mybrand, 2017-02-21:
--db-automigrate option is work!!
Thanks.

closed 2017-02-21 by mybrand