brocaar/chirpstack-application-server #24: Data is not decrypted in MQTT

bramkennis, 2017-02-21:
When subscribing to the tx topic from a particular sensor the data is encrypted.
example:

{
  "applicationID" : "1",
  "applicationName" : "Test",
  "nodeName" : "TesT",
  "devEUI" : "0004a30b00196799",
  "rxInfo" : [ {
    "mac" : "008000000000b6a9",
    "time" : "2017-02-21T09:50:58.195271Z",
    "rssi" : -63,
    "loRaSNR" : 11.2
  } ],
  "txInfo" : {
    "frequency" : 868100000,
    "dataRate" : {
      "modulation" : "LORA",
      "bandwidth" : 125,
      "spreadFactor" : 8
    },
    "adr" : true,
    "codeRate" : "4/5"
  },
  "fCnt" : 54,
  "fPort" : 2,
  "data" : "ESJxAyAAAyABI0VjVkiIgQ=="
}

brocaar, 2017-02-21:
Without knowing the input, it is hard to judge if the data is encrypted or not 😉 You might mix up encoding vs encryption. The data should be decrypted, but is presented in base64 encoding (to make the binary data printable as ASCII). See: https://play.golang.org/p/JNIMHYKz6m (for base64 decoding your data) and https://en.wikipedia.org/wiki/Base64.

bramkennis, 2017-02-21:
You are right. When using the website https://play.golang.org we get the HEX value of our sensors.
Now the programmer must listen to the mqtt topic en decode the data so that we can store the ascii value in db and create some graphs.
Thanks

closed 2017-02-21 by brocaar