brocaar/chirpstack-application-server #25: Data not received with MQTT !!

AnisBH, 2017-02-21:
Hi Bocaar :) , 

Please i have a problem, i can't receive data with MQTT , the device join network  OTAA without any problems then send data is successful but to see data with moquitto_sub  doesn't work :/  please where is exactly the problem? 

Thanks Bocaar :)

AnisBH, 2017-02-21:
Hello :) , 

the problem that i'm using this command  [AppEUI = 70B3D57EF000391A ] :  
mosquitto_sub -t "application/70B3D57EF000391A/#" -v

but when i use this command : 
mosquitto_sub -v -t "application/#"

i receiv data , and i see as Application ID  => application/1/node/0004a30b001ae41e/rx {"applicationID":"1", ... 

Please , applicationID is not 70B3D57EF000391A ? why i receive "1" ?

brocaar, 2017-02-21:
Note that `AppEUI` is not `ApplicationID`. I'll update the docs / UI to make this more clear.

Multiple nodes with different `AppEUI` values could still serve the same application / purpose (e.g. weather station). The `ApplicationID` matches the ID of the application that you created, before adding any nodes. See also the `/api/applications?limit=10` api endpoint :-)

AnisBH, 2017-02-23:
Thanks broccar :) Thank you for your work and your efforts . 
Please clear us this point on the doc . 
Regards ,

commit f3939f4b (2017-02-23): Clarify application ID.

Closes #25.
closed 2017-02-28 by brocaar