brocaar/chirpstack-application-server #26: Uplink frame-counter changed small，when refresh the web

Deepdone, 2017-02-22:
Refresh the page, the Uplink frame-counter just before is 1893, after a moment is 1891, then 1892 at last
[image]

brocaar, 2017-02-22:
Could you describe how I could replicate this behaviour? At my side the Uplink frame-counter increments each time when I receive an uplink frame, which is as expected.

Do you see the same behaviour when looking at these values, but then using the `/api` endpoint? Just to make sure you're not accidentally increasing / decreasing this value by scrolling up and down (e.g. some browsers will automatically increase / decrease nummeric fields when you focus on these fields).

Deepdone, 2017-02-23:
@brocaar  now it's work well.I guess the reason is : my net is not good! when ping the router,the delay is too high; thanks U :100:

closed 2017-02-23 by Deepdone