brocaar/chirpstack-application-server #30: Starting the web interface

MohammedBENHADINE, 2017-03-01:
Hello everyone,
first of all thanks a lot for your great work,
I'm using a windows environment to set up the LoRaWAN backend,
So,i tried to use the CloudMQTT as a mqtt broker.The Bridge is working nicely and it handles to state of my Microchip Gateway.
For the two other element (loraserver and lora-app-server) i can't open the "inventory"interface.
I used the command to create a tls certification and key then i put them in the both directory of the loraserver and the lora-app-server.
When i try to access from my browser to the localhost:8080,i get a TLS handshake error in the lora-app-server line command.
I think i'm mistaking something somewhere.
Thanks in advance for any help
Mohammed.

brocaar, 2017-03-01:
Try again, but with `https://` in front of the url 😉

MohammedBENHADINE, 2017-03-01:
What a beautiful interface !
Thanks a lot Boss ;)

closed 2017-03-01 by MohammedBENHADINE

hpcdan, 2017-12-01:
I am having the same issue, but I am starting with https://localhost:8080

I get the dreaded TLS handshake error.  I'm running the standard docker image, which I assume is (based on the Dockerfile) building the TLS certificates.  Is that an incorrect assumption?

This happens as soon as I try to create a network server

appserver_1      | time="2017-12-01T17:12:15Z" level=info msg="migrations applied" count=21 
appserver_1      | time="2017-12-01T17:12:15Z" level=info msg="starting application-server api" bind="0.0.0.0:8001" ca-cert= tls-cert= tls-key= 
appserver_1      | time="2017-12-01T17:12:15Z" level=info msg="starting join-server api" bind="0.0.0.0:8003" ca_cert= tls_cert= tls_key= 
appserver_1      | time="2017-12-01T17:12:15Z" level=info msg="starting client api server" bind="0.0.0.0:8080" tls-cert="/etc/lora-app-server/certs/http.pem" tls-key="/etc/lora-app-server/certs/http-key.pem" 
appserver_1      | time="2017-12-01T17:12:15Z" level=info msg="registering rest api handler and documentation endpoint" path="/api" 
**appserver_1      | 2017/12/01 17:08:27 http: TLS handshake error from 127.0.0.1:51476: tls: first record does not look like a TLS handshake
appserver_1      | time="2017-12-01T17:08:27Z" level=info msg="transport: http2Client.notifyError got notified that the client transport was broken read tcp 127.0.0.1:51476->127.0.0.1:8080: read: connection reset by peer."**

brocaar, 2017-12-01:
@hpcdan I just tried to reproduce your issue using https://docs.loraserver.io/install/docker/

This first time I loaded the web-interface I got:

```
appserver_1      | 2017/12/01 17:23:57 http: TLS handshake error from 172.18.0.1:39270: EOF
appserver_1      | 2017/12/01 17:23:57 http: TLS handshake error from 172.18.0.1:39272: EOF
appserver_1      | 2017/12/01 17:23:57 http: TLS handshake error from 172.18.0.1:39274: EOF
appserver_1      | 2017/12/01 17:23:57 http: TLS handshake error from 172.18.0.1:39276: EOF
appserver_1      | 2017/12/01 17:23:57 http: TLS handshake error from 172.18.0.1:39278: EOF
appserver_1      | 2017/12/01 17:23:57 http: TLS handshake error from 172.18.0.1:39280: EOF
appserver_1      | 2017/12/01 17:23:57 http: TLS handshake error from 172.18.0.1:39282: EOF
```

However, this was because an other server was running with a different certificate on the same IP and port and this certificate was still in the cache. A clear cache + refresh did the trick.

In case your issue persists, please refer to https://forum.loraserver.io, it is a better place for support questions :-)

xpablobri, 2020-09-21:
borca

> @hpcdan I just tried to reproduce your issue using https://docs.loraserver.io/install/docker/
>
> This first time I loaded the web-interface I got:
>
> ```
> appserver_1      | 2017/12/01 17:23:57 http: TLS [...]

@brocaar pls, can you help me, i dont use a rak, im using only a tranceptor conected to rb3, i installed this https://github.com/RAKWireless/rak_common_for_gateway , to have global and local_json, i configured this to work in 433 Mhz, and i configured gateway, network, and application chirpstackservers, and i used https://github.com/tftelkamp/single_chan_pkt_fwd to and in the interface i can watch "last seen:a few seconds ago" but, in the application i watch "last seen:never" i need a solution pls, how i can watch applicaiton?? pls