brocaar/chirpstack-application-server #35: How to view the historical data of uplink in database

Deepdone, 2017-03-17:
hi @brocaar ，How to view the historical data of uplink in database？I want to show all the uplink data of a node on the web page.Tks

brocaar, 2017-03-19:
This is currently not possible. All data is published to the MQTT topics as mentioned in the docs. This might be a feature for the future, but currently you would have to create a MQTT handler and store the data yourself.

pmiam999, 2017-04-13:
@Deepdone : You can access the systemd log file on the bridge and in case you want to see the packet content, you can use node.js module lora-packet.
@Brocaar: would be great to add a syslog part in the UI to search for such logs files.

The UI is turning into a global management UI (like Cisco FDN since you added gateway mgmt), would make sense to crawl in the logs too.

brocaar, 2017-04-25:
Would #61 solve your question @Deepdone ?

@pmiam999 not sure about your feature request regarding displaying logs in LoRa App Server. Logging is really something managed by the user. One might discard it, one might store it in a centralised log server (eg. ELK stack), some might write it to a file. A lot of variations are possible.

Deepdone, 2017-04-26:
@brocaar i used the InfluxDB to store the history data;and it work well.

RogerioCassares, 2017-06-01:
hello @Deepdone , please let me know about you project, please? I'm having some trouble to find the log-files in gateway bridge and have downloaded to an data base and have this options saved on my application, please!

Best Regard
Rogerio

closed 2017-07-15 by brocaar