brocaar/chirpstack-application-server #36: Fail to make requirements ui-requirements

tamagohan, 2017-03-17:
Hi, @brocaar

I try to build according to [this procedure](https://github.com/brocaar/lora-app-server#building-from-source) on docker.
However, `make requirements ui-requirements` command got following error
```
root@df1b6f81503b:/go/src/github.com/brocaar/lora-app-server# make requirements ui-requirements
Installing development tools
Installing UI requirements
/bin/sh: 1: npm: not found
Makefile:64: recipe for target 'ui-requirements' failed
make: *** [ui-requirements] Error 127
```

So, I modified `Dockerfile` as following and retry.
[Dockerfile]
```
RUN apt-get update
RUN apt-get -y install npm nodejs
RUN ln -s /usr/bin/nodejs /usr/bin/node
```
`make requirements ui-requirements` command succeeded.
However, `make ui statics` command got following error.

```
root@2c35fa3469fe:/go/src/github.com/brocaar/lora-app-server# make ui statics
Building ui

> ui@0.1.0 build /go/src/github.com/brocaar/lora-app-server/ui
> react-scripts build

/go/src/github.com/brocaar/lora-app-server/ui/node_modules/react-scripts/node_modules/cross-spawn/index.js:34
            throw new Error(
                  ^
Error: In order to use spawnSync on node 0.10 or older, you must install spawn-sync:

  npm install spawn-sync --save
    at Function.spawnSync [as sync] (/go/src/github.com/brocaar/lora-app-server/ui/node_modules/react-scripts/node_modules/cross-spawn/index.js:34:19)
    at Object.<anonymous> (/go/src/github.com/brocaar/lora-app-server/ui/node_modules/react-scripts/bin/react-scripts.js:22:22)
    at Module._compile (module.js:456:26)
    at Object.Module._extensions..js (module.js:474:10)
    at Module.load (module.js:356:32)
    at Function.Module._load (module.js:312:12)
    at Function.Module.runMain (module.js:497:10)
    at startup (node.js:119:16)
    at node.js:906:3

npm ERR! ui@0.1.0 build: `react-scripts build`
npm ERR! Exit status 8
npm ERR!
npm ERR! Failed at the ui@0.1.0 build script.
npm ERR! This is most likely a problem with the ui package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     react-scripts build
npm ERR! You can get their info via:
npm ERR!     npm owner ls ui
npm ERR! There is likely additional logging output above.
npm ERR! System Linux 4.4.17-boot2docker
npm ERR! command "/usr/bin/nodejs" "/usr/bin/npm" "run" "build"
npm ERR! cwd /go/src/github.com/brocaar/lora-app-server/ui
npm ERR! node -v v0.10.29
npm ERR! npm -v 1.4.21
npm ERR! code ELIFECYCLE
npm WARN This failure might be due to the use of legacy binary "node"
npm WARN For further explanations, please read
/usr/share/doc/nodejs/README.Debian

npm ERR!
npm ERR! Additional logging details can be found in:
npm ERR!     /go/src/github.com/brocaar/lora-app-server/ui/npm-debug.log
npm ERR! not ok code 0
Makefile:36: recipe for target 'ui' failed
make: *** [ui] Error 1
```

brocaar, 2017-03-19:
Thanks for reporting! Would you be able to do some testing with https://github.com/creationix/nvm and installing the latest NodeJS version?

aMasakiTakahashi, 2017-03-21:
Thank you for reply, @brocaar 

I add some commands to [`Dockerfile`](https://github.com/brocaar/loraserver/blob/master/Dockerfile) to use `nvm`.
```
RUN apt-get update
RUN apt-get -y install build-essential libssl-dev
RUN curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.33.1/install.sh | bash
ENV NVM_DIR "/root/.nvm"
RUN /bin/bash -c "source $NVM_DIR/nvm.sh; nvm install node; nvm use node"

RUN apt-get -y install ruby-dev ruby
RUN gem install fpm
```
(I added commands to install ruby, because `make package-deb` command required `fpm` gem)

In an environment where node was installed by nvm, `make ui statics` command succeeded!
(Installed node version is `v7.7.3`)

Thank you!

tamagohan, 2017-03-21:
Thank you for reply, @brocaar 

I add some commands to [`Dockerfile`](https://github.com/brocaar/loraserver/blob/master/Dockerfile) to use `nvm`.
```
RUN apt-get update
RUN apt-get -y install build-essential libssl-dev
RUN curl -o- https://raw.githubusercontent.com/creationix/nvm/v0.33.1/install.sh | bash
ENV NVM_DIR "/root/.nvm"
RUN /bin/bash -c "source $NVM_DIR/nvm.sh; nvm install node; nvm use node"

RUN apt-get -y install ruby-dev ruby
RUN gem install fpm
```
(I added commands to install ruby, because `make package-deb` command required `fpm` gem)

In an environment where node was installed by nvm, `make ui statics` command succeeded!
(Installed node version is `v7.7.3`)

Thank you!

tamagohan, 2017-03-24:
Hi, @brocaar 
I create [PR](https://github.com/brocaar/lora-app-server/pull/44) to fix build failures.
Please check.

brocaar, 2017-03-24:
Cheers, I merged your changes!

closed 2017-03-24 by brocaar

dreamerzjy, 2017-05-05:
Hi,i got error like this:
Building ui

> ui@0.1.0 build /root/loraserver/src/github.com/brocaar/lora-app-server/ui
> react-scripts build

Creating an optimized production build...
Failed to compile.

Module not found: Error: Cannot resolve module 'create-react-class' in /root/loraserver/src/github.com/brocaar/lora-app-server/ui/node_modules/react-input-autosize/lib

npm ERR! Linux 3.16.0-30-generic
npm ERR! argv "/root/.nvm/versions/node/v7.7.3/bin/node" "/root/.nvm/versions/node/v7.7.3/bin/npm" "run" "build"
npm ERR! node v7.7.3
npm ERR! npm  v4.1.2
npm ERR! code ELIFECYCLE
npm ERR! ui@0.1.0 build: `react-scripts build`
npm ERR! Exit status 1
npm ERR! 
npm ERR! Failed at the ui@0.1.0 build script 'react-scripts build'.
npm ERR! Make sure you have the latest version of node.js and npm installed.
npm ERR! If you do, this is most likely a problem with the ui package,
npm ERR! not with npm itself.
npm ERR! Tell the author that this fails on your system:
npm ERR!     react-scripts build
npm ERR! You can get information on how to open an issue for this project with:
npm ERR!     npm bugs ui
npm ERR! Or if that isn't available, you can get their info via:
npm ERR!     npm owner ls ui
npm ERR! There is likely additional logging output above.

npm ERR! Please include the following file with any support request:
npm ERR!     /root/loraserver/src/github.com/brocaar/lora-app-server/ui/npm-debug.log
make: *** [ui/build] Error 1

the log is:
root@loraserver:~/loraserver/src/github.com/brocaar/lora-app-server# cat /root/loraserver/src/github.com/brocaar/lora-app-server/ui/npm-debug.log
0 info it worked if it ends with ok
1 verbose cli [ '/root/.nvm/versions/node/v7.7.3/bin/node',
1 verbose cli   '/root/.nvm/versions/node/v7.7.3/bin/npm',
1 verbose cli   'run',
1 verbose cli   'build' ]
2 info using npm@4.1.2
3 info using node@v7.7.3
4 verbose run-script [ 'prebuild', 'build', 'postbuild' ]
5 info lifecycle ui@0.1.0~prebuild: ui@0.1.0
6 silly lifecycle ui@0.1.0~prebuild: no script for prebuild, continuing
7 info lifecycle ui@0.1.0~build: ui@0.1.0
8 verbose lifecycle ui@0.1.0~build: unsafe-perm in lifecycle true
9 verbose lifecycle ui@0.1.0~build: PATH: /root/.nvm/versions/node/v7.7.3/lib/node_modules/npm/bin/node-gyp-bin:/root/loraserver/src/github.com/brocaar/lora-app-server/ui/node_modules/.bin:/root/.nvm/versions/node/v7.7.3/bin:/usr/local/sbin:/usr/local/bin:/usr/sbin:/usr/bin:/sbin:/bin:/usr/games:/usr/local/games:/usr/local/go/bin:/root/loraserver/bin:/usr/local/liteide/bin:/usr/local/go/bin:/root/loraserver/bin:/usr/local/liteide/bin
10 verbose lifecycle ui@0.1.0~build: CWD: /root/loraserver/src/github.com/brocaar/lora-app-server/ui
11 silly lifecycle ui@0.1.0~build: Args: [ '-c', 'react-scripts build' ]
12 silly lifecycle ui@0.1.0~build: Returned: code: 1  signal: null
13 info lifecycle ui@0.1.0~build: Failed to exec build script
14 verbose stack Error: ui@0.1.0 build: `react-scripts build`
14 verbose stack Exit status 1
14 verbose stack     at EventEmitter.<anonymous> (/root/.nvm/versions/node/v7.7.3/lib/node_modules/npm/lib/utils/lifecycle.js:279:16)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at EventEmitter.emit (events.js:194:7)
14 verbose stack     at ChildProcess.<anonymous> (/root/.nvm/versions/node/v7.7.3/lib/node_modules/npm/lib/utils/spawn.js:40:14)
14 verbose stack     at emitTwo (events.js:106:13)
14 verbose stack     at ChildProcess.emit (events.js:194:7)
14 verbose stack     at maybeClose (internal/child_process.js:899:16)
14 verbose stack     at Process.ChildProcess._handle.onexit (internal/child_process.js:226:5)
15 verbose pkgid ui@0.1.0
16 verbose cwd /root/loraserver/src/github.com/brocaar/lora-app-server/ui
17 error Linux 3.16.0-30-generic
18 error argv "/root/.nvm/versions/node/v7.7.3/bin/node" "/root/.nvm/versions/node/v7.7.3/bin/npm" "run" "build"
19 error node v7.7.3
20 error npm  v4.1.2
21 error code ELIFECYCLE
22 error ui@0.1.0 build: `react-scripts build`
22 error Exit status 1
23 error Failed at the ui@0.1.0 build script 'react-scripts build'.
23 error Make sure you have the latest version of node.js and npm installed.
23 error If you do, this is most likely a problem with the ui package,
23 error not with npm itself.
23 error Tell the author that this fails on your system:
23 error     react-scripts build
23 error You can get information on how to open an issue for this project with:
23 error     npm bugs ui
23 error Or if that isn't available, you can get their info via:
23 error     npm owner ls ui
23 error There is likely additional logging output above.
24 verbose exit [ 1, true ]

dreamerzjy, 2017-05-05:
i fixed this problem with "npm install  create-react-class"