brocaar/chirpstack-application-server #37: Changing ABP session parameters not working

iBrick, 2017-03-17:
Hi, In last 2 releases (starting from version 0.5.0) there's a bug with updating ABP session parameters.
When no ABP parameters are used you may set them and it's OK, but afterwards if you want to change it to another values there's an error message:
"Error rpc error: code = 13 desc = transport is closing (code: 2)"

brocaar, 2017-03-17:
Ai, that should not happen and should be fixed asap. I'll check asap! Thanks for reporting.

brocaar, 2017-03-17:
I've found the issue and working on a fix.

brocaar, 2017-03-18:
@iBrick thanks again for reporting this issue! I've fixed this in [LoRa Server 0.15.1](https://github.com/brocaar/loraserver/releases/tag/0.15.1). Binaries will be available in a couple of minutes.

closed 2017-03-18 by brocaar

iBrick, 2017-03-20:
Hi @brocaar ,
Sorry to say but now it's even worse. 
With new version I can't even create ABP session settings when there's nothing in the field. 
Pressing 'generate' button or adding them manually gives error: "Error grpc: the connection is unavailable (code: 14)"

brocaar, 2017-03-20:
Are you sure the LoRa Server service has been started?

iBrick, 2017-03-20:
Ooops, no it's not (
Sorry for wrong issue

brocaar, 2017-03-20:
No worries :-)