brocaar/chirpstack-application-server #38: Go build fails

muka, 2017-03-21:
Hi 
I am trying to run from master both docker compose and go build but those import are unreachable
```
	"github.com/brocaar/lora-app-server/internal/migrations"
	"github.com/brocaar/lora-app-server/internal/static"
```

```
main.go:36:2: cannot find package "github.com/brocaar/lora-app-server/internal/migrations" in any of:
...
main.go:37:2: cannot find package "github.com/brocaar/lora-app-server/internal/static" in any of:
...
```

brocaar, 2017-03-21:
You have to run `make statics` first. See the `README` (these commands were updated recently).

muka, 2017-03-21:
thanks, sorry!

muka, 2017-03-21:
Probably this should be updated in the Makefile under `build`

brocaar, 2017-03-21:
Actually that is a good one as the Makefile can detect if these folders exists, if not run these steps.

muka, 2017-03-21:
Hello there,
I made a container to build the ui and inject in the statics
https://github.com/brocaar/lora-app-server/compare/master...muka:ui_build

This may allow docker-compose to run without external deps (=node)

brocaar, 2017-03-24:
@muka would you be able to test this branch? https://github.com/brocaar/lora-app-server/tree/improve_make

See https://github.com/brocaar/lora-app-server/pull/45/files for the changes I've made.

Please note that there were some changes in the `Dockerfile` too, so you might need to rebuild your container so that all requirements are installed (npm was added to build the ui).

muka, 2017-03-27:
Uhm.. I  cannot make it work from compose (but  plain `make *` work perfectly on my pc)

```
# docker-compose up --build appserver
... docker build ...
appserver_1      | Building ui
appserver_1      | /bin/sh: 1: npm: not found
appserver_1      | make: *** [ui/build] Error 127
appserver_1      | Makefile:37: recipe for target 'ui/build' failed
loraappserver_appserver_1 exited with code 2
```

Seems like nvm is not setting npm in the $PATH or something similar

ciudadoscura, 2017-04-12:
I was getting similar issues. The problem was I haven't a large enough swap file. Now the `make requirements ui-requirements` command works perfect.
You can check if you have a swap file by issuing:
`swapon -s`

If not, then issue something like:
`sudo dd if=/dev/zero of=/swapfile bs=1024 count=1024k`
`sudo mkswap /swapfile`
`sudo swapon /swapfile`

Then check the swap file again:
`swapon -s`

brocaar, 2017-04-25:
UI + statics are now automatically compiled and generate on `make build`.

closed 2017-04-25 by brocaar