brocaar/chirpstack-application-server #39: JWT token for the REST API interactive console

ymkins, 2017-03-21:
The REST API interactive console wont work since 0.6.0 update.

Response Body
```
{
  "error": "authentication failed: get token from context error: no authorization-data in metadata",
  "code": 16
}
```
Seems like have to set the Grpc-Metadata-Authorization header field according to the documentation.

brocaar, 2017-03-21:
Note that you can past a JWT token in the console (right top). To generate one, you could use the `/api/internal/login` endpoint (from the api console).

closed 2017-03-21 by ymkins

AnnaLupanova, 2021-07-22:
I am having the same problem. I inserted the token into the upper corner. But doesn't work

NelsonKon, 2021-09-22:
I am having the same problem. I inserted the token into the upper corner. But doesn't work.  What could be the problem?