brocaar/chirpstack-application-server #40: Node session information disappeared in web-interface

samng6, 2017-03-22:
Good day,

Yesterday my colleague has registered two nodes into Lora server with ABP protocol. We entered the Network session key and Application session key. Today when we visited the web interface again, the node session information of one of the nodes has been disappeared. However, we can find the Network session key and Application session key of that node in PostgreSQL. We are wondering whether the node session is expired. In this situation, do we need to register the nodes again? Waiting for your answer.

Best Regards,
Sam

brocaar, 2017-03-23:
Hi @samng6 it is a bit hard to make any conclusions based on this information.

Note that there are two components:

- LoRa Server: stores node-sessions in Redis (which will eventually expire)
- LoRa App Server: stores node config in PostgreSQL (which will never expire)

LoRa App Server stores `NwkSKey` and `AppSKey` in PostgreSQL, but this does not mean the session is still active, as it might already be expired (in LoRa Server).

samng6, 2017-03-24:
Just to be sure, does this mean that it is expected behavior for all ABP sessions to eventually expire?  If this is the case, is only OTAA considered to be usable long-term?

brocaar, 2017-03-24:
Currently yes, see also: https://github.com/brocaar/loraserver/issues/100.

samng6, 2017-03-24:
Thanks for your answer. Does it mean that the node session will be expired after 5 days of inactivity?

closed 2017-07-15 by brocaar