brocaar/chirpstack-application-server #41: Error authentication failed:

helloworld8686, 2017-03-22:
https://192.168.1.165:8080

Error authentication failed: get token from context error: no authorization-data in metadata (code: 16)

jcourtat, 2017-03-22:
Same for me. Installed on Centos 7, without docker. 
make test is successful.
loraserver running at the same time.

Using a168a676f3c924a9ba36ee5b3509cd38407e7f93 (HEAD)
invoked with build/lora-app-server --postgres-dsn postgres://loraserver:dbpassword@localhost/loraserver?sslmode=disable  --http-tls-cert /home/jcourtat/CA/servercert.pem --http-tls-key /home/jcourtat/CA/serverkey.pem
INFO[0000] starting LoRa App Server                      docs="https://docs.loraserver.io/" version=0.6.0
INFO[0000] connecting to postgresql                     
INFO[0000] setup redis connection pool                  
INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://localhost:1883"
INFO[0000] connecting to network-server api              ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
INFO[0000] starting application-server api               bind="0.0.0.0:8001" ca-cert= tls-cert= tls-key=
INFO[0000] starting client api server                    bind="0.0.0.0:8080" tls-cert="/home/jcourtat/CA/servercert.pem" tls-key="/home/jcourtat/CA/serverkey.pem"
INFO[0000] handler/mqtt: connected to mqtt broker       
INFO[0000] handler/mqtt: subscribling to tx topic        topic="application/+/node/+/tx"
INFO[0000] registering rest api handler and documentation endpoint  path="/api"

brocaar, 2017-03-22:
Did you clear your browser cache? Note that on installation / on upgrade to 0.6.0 a default admin is being created with `admin:admin` as username:password.

jcourtat, 2017-03-22:
Correct me if i'm wrong,
My understanding is i can log in with a browser to https://localhost:8080/ without the JWT. i just have to use the admin:admin credential.
If i want to use the JSON RESTful API, i need to use the JWT.
I cleared the firefox cache and restarted both loraserver and lora-app-server, added the site (localhost) in exception list of firefox, but i still can't log in.
here are commands i did to make the lora-app-server:
 make requirements ui-requirements
 make ui statics
 make build
 TEST_POSTGRES_DSN=postgres://loraserver:dbpassword@localhost/loraserver?sslmode=disable make test

Maybe i forget something ?

jcourtat, 2017-03-22:
build/lora-app-server --postgres-dsn postgres://loraserver:dbpassword@localhost/loraserver?sslmode=disable  --http-tls-cert /home/jcourtat/CA/servercert.pem --http-tls-key /home/jcourtat/CA/serverkey.pem --jwt-secret lorawan

curl -k -X GET --header 'Accept: application/json' --header 'Grpc-Metadata-Authorization: eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJsb3JhLWFwcC1zZXJ2ZXIiLCJhdWQiOiJsb3JhLWFwcC1zZXJ2ZXIiLCJuYmYiOjE0OTAxOTI5NjEsImV4cCI6MTQ5OTY1MzM1OCwic3ViIjoidXNlciIsInVzZXJuYW1lIjoiYWRtaW4ifQ.PizGOz6rl2GvhZ0a1tu1ayphzQuS1Yg2CHNBg5LJ9vY' 'https://127.0.0.1:8080/api/applications'
{"error":"authentication failed: not authorized","code":16}

Maybe it would be related to the fact i didn't installed debian package but compiled and run by myself ?

jcampanell-cablelabs, 2017-03-22:
The UI login process generates a JWT behind the scenes that gets put into the header for subsequent requests from the UI.  So if the UI works, JWT validation is working.  If not, then something else is wrong.  Perhaps the admin user was somehow not provisioned?  Did you run lora-app-server with the --db-automigrate switch to update the database?  (This only has to be done with each new version once, though it is minimal startup overhead to do it all the time - so you don't forget like I have a couple of times.)

I just gave your curl command a try on my self-built lora-app-server, and it worked fine.  How did you get your token?

My process: I started Chrome and accessed the UI, accepting the self-signed certificate.  I then got my own JWT token by logging in using Chrome's Postman app.  I then cut-n-pasted your curl command above, replacing your JWT with my own that I got from Postman.  Got my expected {"totalCount":"1"}  response (add limit and offset to the query to get application details - but that did not affect the authorization).

Hope that helps.

jcourtat, 2017-03-23:
Hello,
thank you for your time spent in testing my curl command.
I didn't run the lora-app-server with --db-automigrate, so i did it but unfortunately it didn't fix anything.
I'm stuck with both the ui and the curl. i tryed with both chrome and firefox, but the result is the same.
Here is the HTTP data of the login:
**General**
  Request URL:https://localhost:8080/api/internal/login
  Request Method:POST
  Status Code:401 
  Remote Address:[::1]:8080

**Response Headers**
  content-length:50
  content-type:application/json
  date:Thu, 23 Mar 2017 09:37:12 GMT
  grpc-metadata-trailer:Grpc-Status
  grpc-metadata-trailer:Grpc-Message
  status:401

**Request Headers**
  :authority:localhost:8080
  :method:POST
  :path:/api/internal/login
  :scheme:https
  accept:*/*
  accept-encoding:gzip, deflate, br
  accept-language:en-GB,en-US;q=0.8,en;q=0.6
  content-length:39
  content-type:text/plain;charset=UTF-8
  origin:https://localhost:8080
  referer:https://localhost:8080/
  user-agent:Mozilla/5.0 (X11; Linux x86_64) AppleWebKit/537.36 (KHTML, like Gecko)   Chrome/56.0.2924.87 Safari/537.36
**Request Payload**
 {username: "admin", password: "admin"}
 password:"admin" username:"admin"

I wonder if the admin account is created in postgredb

jcourtat, 2017-03-23:
loraserver=> SELECT * FROM application_user;
 id | created_at | updated_at | user_id | application_id | is_admin 
----+------------+------------+---------+----------------+----------
(0 rows)

=> if this is the table that handles the user for connecting through ui, then it's strange there is no admin.
Would it be related to the make test that seems to reset the database ?
Or maybe i need to run another command to create this admin

jcourtat, 2017-03-23:
that's it. i did a make clean, drop the database and role, then recreated it. after make build, i can log in ui with firefox. Maybe the make test deletes too much things. Since i'm not a Go programmer, it's hard to propose a patch to fix this.

brocaar, 2017-03-23:
@jcourtat actually yes, the tests are resetting the database for every test executed. I would recommend you to run your tests against a different database :-)

closed 2017-03-25 by helloworld8686

waytai, 2017-09-22:
curl -X POST --header 'Content-Type: application/json' --header 'Accept: application/json' -d '{ \ 
   "password": "admin", \ 
   "username": "admin" \ 
 }' 'https://localhost:8080/api/internal/login'

I use this to get jwt , but it returns 

curl: (60) SSL certificate problem: self signed certificate
More details here: http://curl.haxx.se/docs/sslcerts.html

curl performs SSL certificate verification by default, using a "bundle"
 of Certificate Authority (CA) public keys (CA certs). If the default
 bundle file isn't adequate, you can specify an alternate file
 using the --cacert option.
If this HTTPS server uses a certificate signed by a CA represented in
 the bundle, the certificate verification probably failed due to a
 problem with the certificate (it might be expired, or the name might
 not match the domain name in the URL).
If you'd like to turn off curl's verification of the certificate, use
 the -k (or --insecure) option.

what if I am want to get jwt just use https