brocaar/chirpstack-application-server #50: port 8080       Unable to connect

helloworld8686, 2017-03-25:
cmd:
         root@416d013cbef4:/go/src/github.com/brocaar/lora-app-server/build# ./lora-app-server
web:
         https://192.168.1.103:8080/

             Unable to connect      
             Firefox can't establish a connection to the server at 192.168.1.103:8080.
log:
INFO[0000] starting LoRa App Server                      docs="https://docs.loraserver.io/" version=0.6.0
INFO[0000] connecting to postgresql                     
INFO[0000] setup redis connection pool                  
INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://mosquitto:1883"
INFO[0000] connecting to network-server api              ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
INFO[0000] applying database migrations                 
INFO[0000] handler/mqtt: connected to mqtt broker       
INFO[0000] handler/mqtt: subscribling to tx topic        topic="application/+/node/+/tx"
INFO[0000] grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8000: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8000 <nil>} 
INFO[0000] migrations applied                            count=0
INFO[0000] starting application-server api               bind="0.0.0.0:8001" ca-cert= tls-cert= tls-key=
INFO[0000] starting client api server                    bind="0.0.0.0:8080" tls-cert="/etc/lora-app-server/certs/http.pem" tls-key="/etc/lora-app-server/certs/http-key.pem"
INFO[0000] registering rest api handler and documentation endpoint  path="/api"
INFO[0001] grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8000: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8000 <nil>}

brocaar, 2017-03-28:
Are you using Docker? If yes, make sure the port is forwarded to the host.

helloworld8686, 2017-03-28:
cmd:
sudo docker-compose run -p 8080:8080 --rm appserver bash
web:
https://192.168.1.103:8080/
ok!!!!!!!!!!!!!!!
thanks!

closed 2017-03-28 by helloworld8686

inotbiokaza, 2017-05-24:
How to install "User Interface on LoRa App Server" without using "Docker"?