brocaar/chirpstack-application-server #51: Feature Request: Search nodes by tag

ymkins, 2017-03-27:
Is it hard to implement additional field in the node structure and API GET /api/nodes/{tag}?
It allows ability to manage collections of nodes by some tags.

brocaar, 2017-03-28:
Could you share which usecase this would serve?

ymkins, 2017-03-28:
@brocaar, I see few usecases:

1st usecase.
As loraserver user I would like manage node collections by node types. One type can be served by different applications.

2nd usecase.
As network provider I would like manage node collections for clients. Each client can have nodes on different applications.

Next usecase.
As client of network provider I would like manage node collections on my own reason. Each collection can be served by different applications.

Right now to get info about few nodes we have to call API GET /api/nodes/{devEUI} few times. With tag we could get the collection of nodes in one API call. I prefer use API and not direct queries to DB.

The main goal is implementing the business logic separately from loraserver.

ymkins, 2017-03-29:
@brocaar, a few words more.
The tagging feature seems for me as general solution that allows manage collections based on the different properties. The placement, the ownership, some trait can be described with tags and used for not only better UX but statistics and analytics.

brocaar, 2017-11-19:
For the time being, this won't be implemented in LoRa App Server. However the good news is that I'm moving more and more LoRaWAN logic into LoRa Server (currently in the process of refactoring the downlink queue, this will be moved to LoRa Server). With that, it will be fairly easy to implement your own device management dashboard on top of LoRa Server instead of LoRa App Server. Note that in the last release, all the device settings already moved to LoRa Server. See also: https://forum.loraserver.io/t/release-lora-server-0-22-0-lora-app-server-0-14-x-read-carefully/350

closed 2017-11-19 by brocaar