brocaar/chirpstack-application-server #52: Question: Receive/send data over REST API

crdn, 2017-03-28:
Hi brocaar

Thank you very much for sharing your great solution!

I set up your project and it runs without problems. Now I would like to visualize the received sensor data on a website. The easiest way would be to receive and send data over the REST API, because it is plattform independent.

Is there any possibility to get the data over the REST API and if not, what would be the best way? Is it planned to extend the REST API as suggested or not?

Thank you very much!
crdn

pmiam999, 2017-04-13:
@crdn: why don't you setup a MQTT handler that would push radio stats into an influxdb database and then show graphs using grafana. I am doing this and this is really easy to setup.

In case you were talking about seeing the payload, that's application/vendor specific, so it doesn't make sense to handle this in the app-server stage.

crdn, 2017-04-13:
Hi pmiam999

Thank you very much for your hint. As you mentioned, the bigger challenge is the conversion and interpretation of the vendor specific payload. So I set up an ELK Stack and I'm importing the data through MQTT into ElasticSearch. Logstash then helps me to convert the data and Kibana does the visualisation.

I worked with this (as far as I know unsupported) Logstash input https://github.com/widarlein/logstash-input-mqtt.

pmiam999, 2017-04-13:
You should have a look at influxdb/grafana, this is a powerful and easy solution to store timeseries data. The decoding part can be a simple perl script listening to a given MQTT topics en pushing to influx (requires few lines of code).

brocaar, 2017-07-15:
Please follow https://github.com/brocaar/lora-app-server/pull/102, this will allow receiving data by setting up HTTP callback urls. The LoRa App Server format will be supported by MyDevices / Cayenne soon so that you can directly display your data in dashboard.

brocaar, 2017-09-04:
I'll release the HTTP integration this week :-)

closed 2017-09-04 by brocaar

proffalken, 2018-10-30:
Circling back late to this, however there is now an (MQTT "Beat")[https://github.com/nathan-K-/mqttbeat] available for Elasticsearch that will pull data from an MQTT queue into Elasticsearch.

I'm using it successfully with The Things Network, I see no reason why it would fail to work with LoraServer.io.

InfluxDB is excellent (we're InfluxData partners we like it so much!), however if you want to scale out Influx properly (i.e. using more nodes rather than just throwing RAM at the problem!), then it becomes very expensive very quickly.  Elasticsearch on the other hand scales out as well as up, and provides data resiliency "out of the box" as a result. (Full disclosure, we're also Elastic.co partners!)

If we only need metrics as opposed to log/event/metric correlation, then we'll use Influx. If we know that we'll need to scale the back-end storage and take advantage of multi-tier data retrieval speeds etc, then we'll use Elastic.