brocaar/chirpstack-application-server #54: 500 Error API /nodes/{devEUI}/activation error: redigo: nil returned

ymkins, 2017-03-29:
I have a question.

I registered a node and can view it in UI and get it by API. 
But on call to the API /nodes/{devEUI}/activation I get the 500 Error.

```
{
  "devEUI": "d000000000001234",
  "appEUI": "a000000000001234",
  "appKey": "fade0e9e2fab676d60600161d054fade",
  "rxDelay": 0,
  "rx1DROffset": 0,
  "channelListID": "0",
  "rxWindow": "RX1",
  "rx2DR": 0,
  "name": "device_d000000000001234",
  "relaxFCnt": false,
  "adrInterval": 0,
  "installationMargin": 0,
  "description": "d1234",
  "isABP": false,
  "applicationID": "13",
  "isClassC": false,
  "useApplicationSettings": true
}

{
  "error": "rpc error: code = 2 desc = get node-session d000000000001234 error: redigo: nil returned",
  "code": 2
}
```
This node was not joined and so has no session. 

So, the question is:
Should I handle the 500 Error with code 2 or there should be some another response?

brocaar, 2017-03-30:
I think that response could be improved, a 404 would be better :-)

Lately I've improved the error handling at a lot of parts, however since some errors are chained (function calling a function calling a function ....), I might have missed one part.

Thanks for mentioning, I'll look into it!

brocaar, 2017-08-15:
I think this has been fixed a while ago :-)

Getting a non-existing activation now returns:

```
{
  "error": "object does not exist",
  "code": 5
}
```

closed 2017-08-15 by brocaar