brocaar/chirpstack-application-server #55: Error pq: relation "gateway" does not exist (code: 2)

simonnilsson, 2017-03-30:
After upgrading **loraserver** to 0.16.1 and **lora-app-server** to 0.7.0 i get the following error on the "Manage Gateway" page, it appears like the gateway table in the database is not created correctly in the migration.

`Error pq: relation "gateway" does not exist (code: 2)`

brocaar, 2017-03-30:
Please see the `--db-automigrate` / `DB_AUTOMIGRATE` config setting, which will create all tables.

simonnilsson, 2017-03-30:
Hi, thanks for a quick response. I have that option enabled in the configuration and have also tried running it manually.

```
INFO[0000] starting LoRa App Server                      docs="https://docs.loraserver.io/" version=0.7.0
INFO[0000] connecting to postgresql
INFO[0000] setup redis connection pool
INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://localhost:1883"
INFO[0000] connecting to network-server api              ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
INFO[0000] applying database migrations
INFO[0000] handler/mqtt: connected to mqtt broker
INFO[0000] handler/mqtt: subscribling to tx topic        topic="application/+/node/+/tx"
INFO[0000] migrations applied                            count=0
INFO[0000] starting application-server api               bind="0.0.0.0:8001" ca-cert= tls-cert= tls-key=
```

brocaar, 2017-03-30:
Sorry, I meant you need to add that config setting to the LoRa Server config (as LoRa Server will keep track of your gateways and stats).

simonnilsson, 2017-03-30:
Okay, I have enabled it in the loraserver configuration as well, but same result unfortunately.

````
level=info msg="applying database migrations"
level=info msg="migrations applied" count=0
```

brocaar, 2017-03-30:
Would you mind sending me a private message on https://gitter.im/loraserver/loraserver, then we can together go over your config and see if it is a config issue or if there is a bug somewhere.

simonnilsson, 2017-03-30:
The problem was that I used the same database for both loraserver and lora-app-server, changing loraserver to its own database solved the problem, thanks @brocaar

closed 2017-03-30 by simonnilsson