brocaar/chirpstack-application-server #57: About SSL

onlymuxia, 2017-03-30:
I think it's very necessary to notice that SSL Certificates generated by openssl is in a illegal-format.

And for developer it is easy to get a legal SSL Certificates with tool in crypto/tls/generate_cert.go

ymkins, 2017-03-31:
@onlymuxia
What did you mean "illegal-format"?
I know about trouble with providing the subjectAltName on openssl command line. It can be solved with sub-shell command.
Can you please point for openssl defects?

onlymuxia, 2017-04-01:
Using SSL Certificates generated by openssl , I get this error:
       crypto/tls: failed to parse key PEM data

onlymuxia, 2017-04-01:
@ymkins 
Using SSL Certificates generated by openssl , I get this error:
crypto/tls: failed to parse key PEM data

brocaar, 2017-04-01:
@onlymuxia are you sure it is an openssl issue? The post-install script of the `.deb` package uses openssl to generate a self-signed certificate on installation. So far that works great for me and I've used this to install LoRa App Server for many people.

https://github.com/brocaar/lora-app-server/blob/master/packaging/deb/post-install.sh#L65

onlymuxia, 2017-04-01:
@brocaar 
              Im sorry that I try again on another computer and SSL generated by openssl is OK.
              Actually it is my fault And I Will  check my config file again.

closed 2017-04-13 by brocaar