brocaar/chirpstack-application-server #58: Improve: Adding extra json field to db and api rest

claudyus, 2017-04-02:
Hi all,
I'm using this project and I would like to propose a tiny improvement that can consolidate future development and better decoupling loraserver data from end-user interfaces.

The idea is to add a postgresql json field [1] to all DB entries that can be retrieve and set using the rest API.

This can be used to store and retrieve additional and custom informations like ip for the gateways or tags for any device or other informations like tenant or acl for each entry, those informations are only stored "within" the entries and are used by upper-layer of the business logics.

Also #51 can be implemented on top of this feature without implement a custom field for each new request.

Thank you

1. https://www.postgresql.org/docs/current/static/datatype-json.html

zh0uquan, 2017-05-01:
@claudyus hey, did you try to manage to add this feature by your own? I also want to have joined field data for infos of gateway by one request rather than the looping way... If not, maybe I should try to add this feature using golang(although I am not familiar with it)

claudyus, 2017-05-02:
Hi @zh0uquan,
also I'm not really familiar with the golang stack used here. At the moment I manually extended the db and I don't have modified this project but I would like to help you in developing this feature!

brocaar, 2017-11-19:
For the time being, this won't be implemented in LoRa App Server. However the good news is that I'm moving more and more LoRaWAN logic into LoRa Server (currently in the process of refactoring the downlink queue, this will be moved to LoRa Server). With that, it will be fairly easy to implement your own device management dashboard / application-server on top of LoRa Server instead of LoRa App Server. Note that in the last release, all the device settings already moved to LoRa Server. See also: https://forum.loraserver.io/t/release-lora-server-0-22-0-lora-app-server-0-14-x-read-carefully/350

closed 2017-11-19 by brocaar