brocaar/chirpstack-application-server #61: Traffic history and last 10 received payload on App and device level

minggi, 2017-04-04:
It would help if I can see the the last 10 received payloads and a traffic history on App and device level.

JairoRotava, 2018-02-17:
It would be good if the LoraWan frame logs keep something like the last 10 messages. Right now the list is cleared everytime you change the manu on the lora app server. Keepping the last messages helps to debug issues on the nodes.
Sigofx backend and another Lorawan services provides this kind of log.

brocaar, 2018-05-10:
Currently I don't have any plans to support this, as it would also impact the performance (which happened for some people when logging all frames the the database, which is why this moved to live data using web sockets recently).

closed 2018-05-10 by brocaar