brocaar/chirpstack-application-server #63: Failed to start LoRa App Server.

VahrushevSA, 2017-04-07:
After version 0.5.0 I can not run Lora app server. I ask to help, what can be done.

brocaar, 2017-04-07:
Check the logs? Check the release notes about what has changed and what settings needs to be updated?

VahrushevSA, 2017-04-07:
It does not start because of postgresql.
line: POSTGRES_DSN=postgres://loraserver:password@localhost/loraserver?sslmode=disable
If you change to drain: POSTGRES_DSN=postgres://loraserver:password@localhost/loraserver?sslmode=disabl
Then lora app server starts with errors.

VahrushevSA, 2017-04-07:
I have postgresql v 9.6.2

brocaar, 2017-04-13:
> Then lora app server starts with errors.

Which errors? ;-)

VahrushevSA, 2017-04-16:
Thanks Brocaar, figured out.
Put the jwt-secret and lora app server started.

closed 2017-04-17 by brocaar