brocaar/chirpstack-application-server #68: Add the possibility of disabling  the decryption of FRMPayload

iwakidi, 2017-04-14:
Hello!

First of all, I want to thank you for all your efforts leading to this project that has been tremendously valuable to me and to my team!

That being said, we are facing an issue with FRMPayload decryption. We have sensors on the ground that send a non encrypted physichal payload. Moreover, they don't have any AppsKey since they were not conceived to join the network with an OTAA. 

As I understood, whenever you have a ConfirmedDataUp message,  you decrypt the data using the AppsKey of the node as linked here : https://github.com/brocaar/lora-app-server/blob/23ddf5160e10abea2463f11b94e974a967001516/internal/api/application_server.go#L238 .

As some of our sensors dont encrypt the data, taking the "req.Data" as is would be suitable and enough. According to you, would it be a good idea to add a flag per application to deactivate FRM decryption for all its nodes ? 

Of course, all of our future sensors will use OTAA activation and would have a standard way of communicating with the NS. But the old ones are not, and don't we are the only ones in this case. I beleive this is the case for a certain amount of already deployed plateforms. 

Hope we can soon here from you. 
Thanks, Imad.

brocaar, 2017-04-17:
As far as I'm aware, encryption is not optional in the LoRaWAN protocol:

> 4.3.3 MAC Frame Payload Encryption (FRMPayload)
> If a data frame carries a payload, FRMPayload must be encrypted before the message integrity code (MIC) is calculated.
The encryption scheme used is based on the generic algorithm described in IEEE 802.15.4/2006 Annex B [IEEE802154] using AES with a key length of 128 bits.

Unless I missed something, I would like to stay as close to the LoRaWAN protocol as possible. Adding work-arounds is easy, removing them is a pain as people start relying on them.

What you could do as a work-around is decrypt (the plaintext data), so that when LoRa App Server will encrypt it, the actual result is the plaintext data.

iwakidi, 2017-04-18:
You're right, adding a workaround is easier, so we're certainely going for it. 
Thanks a lot for your answer !

closed 2017-04-18 by iwakidi

PhatHub, 2017-06-23:
Hi, I'm using the web-facing application interface, and I found that the node frame logs are not showing decrypted data for the frmPayload... I'm just getting the Base64 encoded version of the encrypted data.
Is there another way to get the unencrypted data as mentioned by iwakidi? (Did he get it through the REST API?) 
I can decrypt it manually in python, but I figured I'd ask in this thread first.
Thanks! :)

brocaar, 2017-06-23:
The frame-logs are (currently) as-is, so as they are received by the gateways. This is wat you want to use for debugging, not for your application. Please see https://docs.loraserver.io/lora-app-server/integrate/data/.