brocaar/chirpstack-application-server #70: Symptom: not all tables are created in the DB

uvoz, 2017-04-20:
Installed the sofware and dependencies using apt-get on ubuntu 16.06 and got everything up and running.

lora-app-server version 0.7.2
loraserver version 0.16.1

The http dashboard of the loraserver however complains that it can not find gateway related tables:
**Error pq: relation "gateway" does not exist (code: 2)**

(happens after running the loraserver process first after installation). Running lora-app-server after does not add tables.

Investigated: When running the lora-app-server process as first after installation, the gateway tables are created but the lora-server related tables are not. This is wat the db looks like in such scenario:

loraserver_as=# \d
                    List of relations
 Schema |         Name         |   Type   |     Owner     
--------+----------------------+----------+---------------
 public | gateway              | table    | loraserver_as
 public | gateway_stats        | table    | loraserver_as
 public | gateway_stats_id_seq | sequence | loraserver_as
 public | gorp_migrations      | table    | loraserver_as
(4 rows)

The lora-app-server will then not start properly and crash.

When running loraserver first, loraserver and the app server will both come up. The gateway tables are however never created:

   List of relations
 Schema |          Name           |   Type   |     Owner     
--------+-------------------------+----------+---------------
 public | application             | table    | loraserver_as
 public | application_id_seq      | sequence | loraserver_as
 public | application_user        | table    | loraserver_as
 public | application_user_id_seq | sequence | loraserver_as
 public | channel_list            | table    | loraserver_as
 public | channel_list_id_seq     | sequence | loraserver_as
 public | downlink_queue          | table    | loraserver_as
 public | downlink_queue_id_seq   | sequence | loraserver_as
 public | gorp_migrations         | table    | loraserver_as
 public | node                    | table    | loraserver_as
 public | user                    | table    | loraserver_as
 public | user_id_seq             | sequence | loraserver_as
(12 rows)

Running lora-app-server after does not trigger the creation of the required tables when the loraserver tables are in place.

brocaar, 2017-04-20:
* Make sure you're using two databases (one for lora-app-server, one for loraserver) and make sure this is correctly configured for both services
* Make sure that you start both services with the `--db-automigrate` / `DB_AUTOMIGRATE` flag set

uvoz, 2017-04-20:
Thanks Bro! Works like a charm with two databases.

uvoz, 2017-04-20:
Where's you [donate] button?

closed 2017-04-20 by uvoz