brocaar/chirpstack-application-server #71: api jwt parse error: token contains an invalid number of segments

carlberg74, 2017-04-21:
Hi,

Using loraserver version: 0.16.0 and lora-app-server version: 0.7.2.
Trying to use the api to access nodes/queues etc.

Running (as from api page).
curl -X GET --header 'Accept: application/json' --header 'Grpc-Metadata-Authorization: UDe2KIQWZ/OdPxrk6xLvLNtd73H3FoV6yImsfA38Ywc=' 'https://192.168.1.121:8080/api/nodes/70B3D5E75E000EB3/queue'
Results in: {"error":"authentication failed: jwt parse error: token contains an invalid number of segments","code":16}

Have tested regenerating the token using: 'openssl rand -base64 32', as described in /etc/default/lora-app-server... but..

any advice?

Cheers
-Anders

brocaar, 2017-04-21:
You're mixing the JWT_SECRET with the JWT token. The first one is a secret to generate JWT tokens. The latter one is a signed JSON object. See also www.jwt.io and https://docs.loraserver.io/lora-app-server/api/#authentication-and-authorization.

carlberg74, 2017-04-21:
yes, I was... sorry.
So for others that are newbie as me..

Here is what I did:
Set the JWT_SECRET in /etc/default/lora-app-server to ex. 'MySecret'
Then restart lora-app-server (or all services).
Go to https://jwt.io/ and replace the word 'secret' with ex. 'MySecret' in the textbox under verify signature section,
and add claim below in the payload data textbox.
{
    "iss": "lora-app-server",
    "aud": "lora-app-server",
    "nbf": 1492778496,
    "exp": 1514678400,
    "sub": "user",
    "username": "admin"
}
and press enter or something and see that the token on the left side is updated.
(this claim will expire 2017-12-30)
Copy the token.
Now go to https://192.168.1.121:8080/api and paste the token on the top text box and then use any of the api below.

The gui don't work with Microsoft Edge, use IE or Chome instead.

closed 2017-04-21 by carlberg74