brocaar/chirpstack-application-server #73: Send Downlink Data with MQTT not working

Movsun, 2017-04-25:
Hello brocaar, 

I try to send downlink message via MQTT. I follow the instruction and publish the json data to "application/[applicationID]/node/[devEUI]/tx" but I didn't see my message in "/api/nodes/{devEUI}/queue". 
If I use HTTP Post to send, It works. 
Thank you in advance,

Movsun

brocaar, 2017-05-03:
Are you sure you replaced the `[applicationID]` and `[devEUI]` with the correct application id and (lowercase) devEUI? Could you also check the lora-server-logs?

brocaar, 2017-05-03:
Just checked, but I don't see any problems with the latest version. When enqueueing data over MQTT, I get the following logs (lora-app-server):

```
INFO[8237] handler/mqtt: data-down payload received      topic="application/1/node/0101010101010101/tx"
INFO[8237] downlink queue item enqueued                  dev_eui=0101010101010101 id=1
```

closed 2017-07-15 by brocaar