brocaar/chirpstack-application-server #74: Node creation form improvements

pmiam999, 2017-04-27:
Hi All contributors,

In order to avoid errors on node creation, i would suggest to add the following : 

x Tooltip (or labels) on IDs :
- DEVEUI / APPEUI : should be in little endian format (lsb)
- APPKEY : should be in big endian format (msb)

x Allow to enter uppercase IDs and trom spaces (for lazy users like me).

Thanks !

brocaar, 2017-05-03:
Yes, I agree the input of HEX data could be improved? Maybe by implementing a HEX input widget for ReactJS?

I don't think that DevEUI / AppEUI should be LSB and AppKey should be MSB. This really depends on  your node. In the current format you can use the field value directly with an RN2483 (Microchip) node, but this does indeed not work with the IMST nodes.

Eventually, I'll improve this, but maybe somebody would want to help out to speed up the development? I think in essence what is needed is:

* ReactJS component for HEX input 
* Configurable byte length (so that it can be re-used for DevAddr, DevEUI, AppEUI, AppKey, GW MAC, ...) + validation
* Toggle for LSB / MSB?

What do you think?

pmiam999, 2017-05-04:
Agree with you, i will submit a PR soon.

brocaar, 2017-05-04:
Great! That is much appreciated. Maybe for the LSB / MSB toggle, you could use the `input-group-addon` class from bootstrap? See: http://getbootstrap.com/css/#forms

camilorojasg, 2017-06-08:
How is this work going? I could collaborate.

pmiam999, 2017-06-16:
@camilorojasg : don't have time to add it. Feel free to implement it !

brocaar, 2018-10-13:
I've made some updates (not yet pushed to github). I would appreciate your feedback on this input:

[image]

* The input is better formatted so that you can clearly see the character pairs `01020304` => `01 02 03 04`
* When editing, the keys are hidden by default (click on the eye icon to view / edit)
* Using the MSB button you can switch to LSB, which will reverse the key input

[image]