brocaar/chirpstack-application-server #77: authentication failed: not authorized but can access to :8080/api

vravoire, 2017-05-02:
I can access to https://localhost:8080/api but when i "try it out!" for exemple GET /api/application
I have "error": "authentication failed: not authorized",
          "code": 16

Everything in https://localhost:8080/ is ok
I look into this issues #41, but it didn't fix 

maybe it's because  grpc.WithPerRPCCredentials method is in the wrong place (i write in google.golang.org/grpc/clientconn.go)

brocaar, 2017-05-02:
You need to generate a JWT token and fill this in the JWT token field (right top corner). See also https://docs.loraserver.io/lora-app-server/api/#authentication-and-authorization.

vravoire, 2017-05-02:
Yes I already do this :
I generate a token, I put the secret in /etc/default/lora-app-server (JWT_SECRET) and i fill the jwt token in the token field.

brocaar, 2017-05-02:
Could you try the `/api/internal/login` endpoint to generate a JWT token? For that token no authentication is needed. After posting your credentials it should give you a JWT token. Then you're able to compare it with the token that you generated yourself.

vravoire, 2017-05-02:
ok it works ! 
Thanks for your answer, you do a great job

closed 2017-05-02 by vravoire

geppoleppo, 2017-05-19:
HI,
i have followed the instruction but doesn't work.
The browser respond: "fetching resource list: undefined; Please wait."

RogerioCassares, 2017-05-19:
Hello @geppoleppo with me happened the same thing. 

But like @brocaar had said, try to generate the JWT token in /api/internal/login with your password and login:

[image]

then update to your folder in your machine at /etc/default/lora-app-server

[image]

Just copy in "JWT token field (right top corner)". DO NOT PRESS ENTER!!!

And just let the token there, and run the commands on api!