brocaar/chirpstack-application-server #78: Why is the downlink queue being purged on Class C messages?

reixd, 2017-05-02:
Hi @brocaar,

I stumbled with instructions in the HandleDownlinkQueueItem function, puring the downlink queue when sending a message to a Class C device.
See the file:  https://github.com/brocaar/lora-app-server/blob/a24ddb41a7b210e465e0deb3008958d2974d28ad/internal/downlink/downlink.go#L68

- Why is the downlink queue being purged on Class C messages?
- Why are the Class C messages not beeing queued, as normal messages?

brocaar, 2017-05-03:
In my understanding, there is no concept of a queue in Class-C, as frames are sent immediately to the node. So when sending 10 frames using Class-C, they would all be sent immediately to the node.

Now for confirmed-data (down), when an ACK is received (from the node) you still want to have the last transmitted frame around (to send an ack notification to the user with the reference of that frame).

reixd, 2017-05-03:
@brocaar what do think of introducing an optional flag to enqueue (Class C) downlink message, like any other message? 

Example:
```
            {
                "reference": abcd1234,
                "confirmed": false,
                "fPort": 42,
                "data": base64.b64encode("Hello World!"),
                "force-enqueue": true             // default := false
            }
```
This flag would then allow frames to be sent one after another to a Class C mote.

As my understanding Class C introduces a *continuous listening mode*, so therefore the mote should be able to receive data all the time, except when it is transmitting. In addition the RX2 downlink only frequency (869.525 Mhz) has a duty cycle of 10%. Thus the NS could send several frames (e.g segmented data) to the mote. So it is useful if the network server (the lora-app-server component) is able to enqueue and schedule the data (tx frame at timestamp+timeOnAir+GuardTime). 

A use case could be firmware update over the air. Or exchanging the crypto keys.

Opinions?
Questions?

brocaar, 2017-05-16:
The only issue I see is with confirmed data down, would LoRa App Server in this case keep broadcasting this frame, until it receives an ACK? Then also some firmwares send an ACK when the node is sending data back (which might be a day later). It is currently undefined in the LoRaWAN specs how a node should handle when it receives a Class-C confirmed data-down. 

This makes the Class-C queue a bit problematic in my opinion. When sending data as an application, you expect that the frame is send immediately. In case of a "hanging confirmed-data down item", this would not be the case.

closed 2017-09-04 by brocaar