brocaar/chirpstack-application-server #79: MIGRATE_NODE_SESSIONS= missing

glederer007, 2017-05-03:
Hello Orne,

Apparently "MIGRATE_NODE_SESSIONS=" is missing from /etc/default/lora-app-server when it is created through "sudo apt-get install lora-app-server". And if it is missing, lora-app-server won't start at all. Could you check it?

Regards,

Gabor

brocaar, 2017-05-03:
This was a really old migration flag and the code has been cleaned up recently, because of that the config flag was removed also. I'll try a clean install myself, but I don't see (yet) why LoRa App Server would not start because of that?

Do you have any error message?

brocaar, 2017-05-03:
Just installing LoRa App Server on a clean (VM) instance, but I can't reproduce this issue...

glederer007, 2017-05-03:
I installed loraserver / lora-app-server on a cloud service (arubacloud) a couple of days ago when I couldn't start lora-app-server. Unfortunately I didn't save the error logs. On Friday I will install another vps as well, I will try to reproduce the error then.

brocaar, 2017-05-03:
Perfect, let's keep this issue open until then.

glederer007, 2017-05-13:
Sorry, I was not able to reproduce the issue. I will close the issue.

closed 2017-05-13 by glederer007