brocaar/chirpstack-application-server #80: Error gwmigrate on fresh database

siscia, 2017-05-03:
Hi,

I am running a fresh installation with a virgin database.

I used the ansible provided in the other repo and adapt it to point to my completely empty database.

However lora-app-server keep writing this error:

```
gwmigrate: list gateways error (will retry): rpc error: code = 2 desc = missing destination name organization_id in *[]gateway.Gateway
```

The logs looks like this for lora-app-server:

```
:16:19Z" level=info msg="starting LoRa App Server" docs="https://docs.loraserver.io/" version=0.8.0
:16:19Z" level=info msg="connecting to postgresql"
:16:19Z" level=info msg="setup redis connection pool"
:16:19Z" level=info msg="handler/mqtt: connecting to mqtt broker" server="tcp://localhost:1883"
:16:19Z" level=info msg="connecting to network-server api" ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
:16:19Z" level=info msg="handler/mqtt: connected to mqtt broker"
:16:19Z" level=info msg="handler/mqtt: subscribling to tx topic" topic="application/+/node/+/tx"
:16:19Z" level=info msg="gwmigrate: migrating gateway data from LoRa Server"
:16:19Z" level=error msg="gwmigrate: list gateways error (will retry): rpc error: code = 2 desc = missing destination name organization_id in *[]gatew
:16:21Z" level=error msg="gwmigrate: list gateways error (will retry): rpc error: code = 2 desc = missing destination name organization_id in *[]gatew
:16:23Z" level=error msg="gwmigrate: list gateways error (will retry): rpc error: code = 2 desc = missing destination name organization_id in *[]gatew
:16:25Z" level=error msg="gwmigrate: list gateways error (will retry): rpc error: code = 2 desc = missing destination name organization_id in *[]gatew
```

And like this for loraserver_

```
54Z" level=info msg="starting LoRa Server" band="EU_863_870" docs="https://docs.loraserver.io/" net_id=010203 version=0.16.1
54Z" level=info msg="setup redis connection pool" url="redis://localhost:6379"
54Z" level=info msg="connecting to postgresql"
54Z" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://localhost:1883"
54Z" level=info msg="connecting to application-server" ca-cert= server="127.0.0.1:8001" tls-cert= tls-key=
54Z" level=info msg="no network-controller configured"
54Z" level=info msg="backend/gateway: connected to mqtt server"
54Z" level=info msg="backend/gateway: subscribing to rx topic" topic="gateway/+/rx"
54Z" level=info msg="backend/gateway: subscribing to stats topic" topic="gateway/+/stats"
54Z" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: ge
54Z" level=info msg="migrated node-sessions to new format" errors=0 migrated=0
54Z" level=info msg="applying database migrations"
54Z" level=info msg="migrations applied" count=0
54Z" level=info msg="starting api server" bind="0.0.0.0:8000" ca-cert= tls-cert= tls-key=
55Z" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: ge
57Z" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: ge
00Z" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: ge
04Z" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: ge
11Z" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: ge
```

It is me missing some step?

Cheers,

brocaar, 2017-05-03:
* Are you using two separate databases (one for lora-app-server and one for loraserver)?
* Are both services running with `DB_AUTOMIGRATE=true`?

siscia, 2017-05-03:
1) No... I am using the same database for both... Am i suppose to use two different db? Of so sorry, I really miss it...

2) Yes

----- Messaggio originale -----

brocaar, 2017-05-03:
Maybe that should be better clarified in the docs :-) But yes, you do need to setup two separate databases as else the migration history gets mixed up (+ tables might collide in the future).

brocaar, 2017-05-04:
Just checked, but in the ansible config there are already two dbs defined. Were you using the latest playbook version? Just seeing how this could be improved or if you had an older playbook version.

siscia, 2017-05-06:
Hi,

I started from the last version but I actually modify it to remove what I didn't understand and make some small changes to fit my environment.

I don't have any experience with ansible but the whole play book was flawless to run :+1: 

:)

Thanks again for your work :)

You may close or if you have any further question...

siscia, 2017-05-06:
No, actually there was a little issues on EC2.

One of the first command, the second one if I remember correctly, is "apt-get update" however, on EC2 it prompt a GUI and ask you to choose something... Honestly I am not sure to remember what was it about, maybe something like the GRUB.

The problem is that this GUI actually blocks the ansible script...

Now sure if I was able to explain myself...

Anyhow the distilled part is "On EC2 it doesn't work" 

Now I am out... If you need more details just ping me even here or via email :)

brocaar, 2017-05-08:
Cheers, I've opened a separate issue for this to validate if this happens more often / if there is a way to avoid this behaviour.

closed 2017-05-08 by brocaar

rafamedinamardones, 2017-05-28:
Hi brocaar........
I send this message from Chile

I'm trying to install ther server but i have the next problem

INFO[0000] starting LoRa App Server                      docs="https://docs.loraserver.io/" version=0.9.1
INFO[0000] connecting to postgresql                     
INFO[0000] setup redis connection pool                  
INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://localhost:1883"
INFO[0000] connecting to network-server api              ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
INFO[0000] handler/mqtt: connected to mqtt broker       
INFO[0000] handler/mqtt: subscribling to tx topic        topic="application/+/node/+/tx"
INFO[0000] grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8000: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8000 <nil>} 
INFO[0000] gwmigrate: migrating gateway data from LoRa Server 
ERRO[0000] gwmigrate: list gateways error (will retry): rpc error: code = 14 desc = grpc: the connection is unavailable 
INFO[0001] grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8000: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8000 <nil>} 

I try to find information about this issue, but with no luck........... can you help me?

Thank you very much

siscia, 2017-05-28:
@rafamedinamardones 

You do not have a loraserver listening on port 8000 of the localhost.

If you want a deeper understading of the whole architecture I would suggest you to refer to the documentation.
If you simply want to play around and get started as soon as possible you can simply run a docker-compose: https://github.com/siscia/loraserver-docker-componse
Please, docker is not the recommended way to run the whole infrastructure, it is only useful for short tests.

rafamedinamardones, 2017-05-28:
thanks so much siscia.

I have already installed 3 items.... Loraserver, lora-gateway, lora-app-sever.... but my problems continue...

I will try to do all over again and read the docs again.

If I have troubles again Iwill let you know...... If I success, I will send to you the problem and the solution.

Regards

siscia, 2017-05-28:
Try to look if those processes are bound to a port.

On linux is something like: `netstat -lntp` 

If everything is running fine you should see that those process are bound to some port.

alexanderplanar, 2017-06-07:
Hello, brocaar! 
I have the same problem with loraserver.
I checked, that Динара Server 2003 Server datetime ДПС

alexanderplanar, 2017-06-07:
Hello, brocaar! 
I have the same problem with loraserver.
I checked, that the loraserver and the loraappserver uses the different databases (loraserver_ns and loraserver_as). It is required to initialize the new databases some way before using? I try to start all applications on the one machine.

jcampanell-cablelabs, 2017-06-07:
Run both of the apps with the --db-automigrate on to get the databases configured correctly.  I just use that flag all the time, as it only takes minimal time at startup to check, and I never miss a database change with an upgrade.

alexanderplanar, 2017-06-07:
Thanks for reply, Ok, I will try tomorrow. But, as I remember, I pass this parameter to command line.

alexanderplanar, 2017-06-08:
Hello!
I've tried to pass --db-automigrate, but it still does not run correctrly.

I created new role and database:
root:/home/admin# **su - postgres**
postgres=# **create role loraserver_test with login password 'dbpassword';**
_CREATE ROLE_
postgres=# **create database loraserver_testdb with owner loraserver_test;**
_CREATE DATABASE_
postgres=# **\q**

command line:
'admin:/home/admin/work/lora$ **./lora-app-server --postgres-dsn postgres://loraserver_test:dbpassword@localhost/loraserver_testdb?sslmode=disable --redis-url redis://localhost:6379 --mqtt-server tcp://localhost:1883 --mqtt-username sammy --mqtt-password password --bind 0.0.0.0:8001 --http-bind 0.0.0.0:8080 --ns-server 127.0.0.1:8000 --log-level 4 --db-automigrate**'

output:
>INFO[0000] starting LoRa App Server                      docs="https://docs.loraserver.io/" version=0.10.0
>INFO[0000] connecting to postgresql
>INFO[0000] setup redis connection pool
>INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://localhost:1883"
>INFO[0000] connecting to network-server api  ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
>INFO[0000] applying database migrations
>INFO[0000] handler/mqtt: connected to mqtt broker
>INFO[0000] handler/mqtt: subscribling to tx topic        topic="application/+/node/+/tx"
>INFO[0002] migrations applied                            count=16
>INFO[0002] gwmigrate: migrating gateway data from LoRa Server
>ERRO[0002] gwmigrate: list gateways error (will retry): rpc error: code = 2 desc = missing destination >name organization_id in *[]gateway.Gateway
>ERRO[0004] gwmigrate: list gateways error (will retry): rpc error: code = 2 desc = missing destination >name organization_id in *[]gateway.Gateway
>ERRO[0006] gwmigrate: list gateways error (will retry): rpc error: code = 2 desc = missing destination >name organization_id in *[]gateway.Gateway

Maybe it is important, I did not use security features (TLS, etc.), and did not even install and setup it,  cuz I think I is not important for debugging.

alexanderplanar, 2017-06-08:
Yeah! It works now, I created new database again, and specified --dbautomigrate when run it first time, and it works. Thanks!