brocaar/chirpstack-application-server #81: I want to call the restful method DownlinkQueue fail

leecxn2012, 2017-05-09:
when I call the GET /api/nodes/{devEUI}/queue method ,

			resource = client.resource("https://192.168.0.22:8080/api/nodes/0202020202020202/queue");
			resource.header("Grpc-Metadata-Authorization",map.get("jwt").toString() );
			
			response = resource.type("application/json;charset=UTF-8").get(ClientResponse.class);

I get the error
{"error":"authentication failed: jwt parse error: token contains an invalid number of segments","code":16}

how to do ?

leecxn2012, 2017-05-09:
I got it ,
should that
response = resource.header("Grpc-Metadata-Authorization",map.get("jwt")).type("application/json;charset=UTF-8").get(ClientResponse.class);

closed 2017-05-11 by brocaar

achrafitou, 2017-06-07:
Hi leecxn2012
I'm trying to do the same thing but having issues with ssl. Did your program work right away?