brocaar/chirpstack-application-server #83: Should MQTT message include appEUI?

ckt1010, 2017-05-11:
Hi,
As we parse sensor data depend on node's application, it need get appEUI at first. So should we add appEUI in MQTT message for convenient? 
Thanks!

brocaar, 2017-05-11:
Please note that the purpose of the `appEUI` is not specify the application of your node (e.g. what it does). From the specs:

> The AppEUI is a global application ID in IEEE EUI64 address space that uniquely identifies
> the entity able to process the JoinReq frame.

Therefore the application entity in LoRa App Server was added. As long as you have full control over your nodes `appEUI`, you could use it the way you want, but this will break when you buy a node where this would not be possible.

You could end up with two nodes:
* with different `appEUI` values doing exactly the same
* with to equal `appEUI` values doing something completely different

Therefore it is better to use the `applicationID`.

ckt1010, 2017-05-11:
Hi brocaar,
Thanks for your quick reply. En... Do you means LoRa alliance can't control this appEUI uniquely? And you think we should abandon appEUI to use applicationID?

brocaar, 2017-05-11:
You said:

> As we parse sensor data depend on node's application

What I mean is that the `appEUI` might not be the right source to decide how to parse your sensor data. When you have everything under your own control, this is fine but I don't think the `appEUI` is there to facilitate this usage. Instead it is used the identify the entity to be able to process the JoinReq frame.

ckt1010, 2017-05-12:
Thanks brocaar, I understand now.

closed 2017-05-12 by ckt1010