brocaar/chirpstack-application-server #88: can't add new application

mikumaycry, 2017-05-18:
Error pq: insert or update on table "application" violates foreign key constraint "application_organization_id_fkey" (code: 2)

Run Lora-App-Server from master branch, Error occurred when adding first application

brocaar, 2017-05-18:
It is very likely that you did not run the database migrations. Please refer to the `migrations` folder in the source repository, or start the `lora-app-server` binary with the `--db-automigrate` flag. Also make sure to use separate databases for LoRa App Server and LoRa Server.

mikumaycry, 2017-05-18:
Test with loraserver release 0.17.0, lora-app-server release 0.9.1.
I create a new postgresql container, add separate roles and corresponding databases following start guide.
Run both service from source code with --db-automigrate flag true.
And error still occurs.

mikumaycry, 2017-05-18:
Sorry, I forget to rebuild the statics and ui resource before run main.go. Make clean and rebuild static resource could solve this issue.

brocaar, 2017-05-19:
Could you close this issue when it has been resolved?

closed 2017-07-15 by brocaar