brocaar/chirpstack-application-server #89: systemd[1]: lora-app-server.service: Main process exited, code=exited, status=1/FAILURE

duy112, 2017-05-18:
tôi gặp lỗi với lora-app-server
tôi cài trên ubuntu xenial
khi tôi  start và check log thì hiện lỗi 
tôi đã config đầy đủ cơ sở dữ liệu
vui lòng giúp đỡ
xin cảm ơn!
log : 
`
ubuntu@ip-172-31-24-151:~$ sudo systemctl start lora-app-server
ubuntu@ip-172-31-24-151:~$ journalctl -u lora-app-server -f -n 50
-- Logs begin at Mon 2017-05-15 17:03:02 UTC. --
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22954]: time="2017-05-18T10:50:11Z" level=info msg="handler/mqtt: subscribling to tx topic" topic="application/+/node/+/tx"
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Main process exited, code=exited, status=1/FAILURE
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Unit entered failed state.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: Stopped LoRa App Server.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: Started LoRa App Server.
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22962]: time="2017-05-18T10:50:11Z" level=info msg="starting LoRa App Server" docs="https://docs.loraserver.io/" version=0.9.1
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22962]: time="2017-05-18T10:50:11Z" level=info msg="connecting to postgresql"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22962]: time="2017-05-18T10:50:11Z" level=info msg="setup redis connection pool"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22962]: time="2017-05-18T10:50:11Z" level=info msg="handler/mqtt: connecting to mqtt broker" server="tcp://localhost:1883"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22962]: time="2017-05-18T10:50:11Z" level=info msg="connecting to network-server api" ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22962]: time="2017-05-18T10:50:11Z" level=info msg="applying database migrations"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22962]: time="2017-05-18T10:50:11Z" level=info msg="handler/mqtt: connected to mqtt broker"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22962]: time="2017-05-18T10:50:11Z" level=info msg="handler/mqtt: subscribling to tx topic" topic="application/+/node/+/tx"
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Main process exited, code=exited, status=1/FAILURE
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Unit entered failed state.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: Stopped LoRa App Server.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: Started LoRa App Server.
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22970]: time="2017-05-18T10:50:11Z" level=info msg="starting LoRa App Server" docs="https://docs.loraserver.io/" version=0.9.1
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22970]: time="2017-05-18T10:50:11Z" level=info msg="connecting to postgresql"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22970]: time="2017-05-18T10:50:11Z" level=info msg="setup redis connection pool"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22970]: time="2017-05-18T10:50:11Z" level=info msg="handler/mqtt: connecting to mqtt broker" server="tcp://localhost:1883"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22970]: time="2017-05-18T10:50:11Z" level=info msg="connecting to network-server api" ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22970]: time="2017-05-18T10:50:11Z" level=info msg="applying database migrations"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22970]: time="2017-05-18T10:50:11Z" level=info msg="handler/mqtt: connected to mqtt broker"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22970]: time="2017-05-18T10:50:11Z" level=info msg="handler/mqtt: subscribling to tx topic" topic="application/+/node/+/tx"
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Main process exited, code=exited, status=1/FAILURE
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Unit entered failed state.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: Stopped LoRa App Server.
May 18 10:50:11 ip-172-31-24-151 systemd[1]: Started LoRa App Server.
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22978]: time="2017-05-18T10:50:11Z" level=info msg="starting LoRa App Server" docs="https://docs.loraserver.io/" version=0.9.1
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22978]: time="2017-05-18T10:50:11Z" level=info msg="connecting to postgresql"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22978]: time="2017-05-18T10:50:11Z" level=info msg="setup redis connection pool"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22978]: time="2017-05-18T10:50:11Z" level=info msg="handler/mqtt: connecting to mqtt broker" server="tcp://localhost:1883"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22978]: time="2017-05-18T10:50:11Z" level=info msg="connecting to network-server api" ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22978]: time="2017-05-18T10:50:11Z" level=info msg="applying database migrations"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22978]: time="2017-05-18T10:50:11Z" level=info msg="handler/mqtt: connected to mqtt broker"
May 18 10:50:11 ip-172-31-24-151 lora-app-server[22978]: time="2017-05-18T10:50:11Z" level=info msg="handler/mqtt: subscribling to tx topic" topic="application/+/node/+/tx"
May 18 10:50:12 ip-172-31-24-151 systemd[1]: lora-app-server.service: Main process exited, code=exited, status=1/FAILURE
May 18 10:50:12 ip-172-31-24-151 systemd[1]: lora-app-server.service: Unit entered failed state.
May 18 10:50:12 ip-172-31-24-151 systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
May 18 10:50:12 ip-172-31-24-151 systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
May 18 10:50:12 ip-172-31-24-151 systemd[1]: Stopped LoRa App Server.
May 18 10:50:12 ip-172-31-24-151 systemd[1]: lora-app-server.service: Start request repeated too quickly.
May 18 10:50:12 ip-172-31-24-151 systemd[1]: Failed to start LoRa App Server.

`

brocaar, 2017-05-19:
Could you try the following:

```bash
# stop the lora-app-server process
systemctl stop lora-app-server

# load configuration
set -a
source /etc/default/lora-app-server

# start loraserver manually
lora-app-server
```

What does that output?

duy112, 2017-05-19:
thanks for help
i've been tried and ouput  : 

ubuntu@ip-172-31-24-151:~$ sudo systemctl stop lora-app-server
ubuntu@ip-172-31-24-151:~$ set -a
ubuntu@ip-172-31-24-151:~$ source /etc/default/lora-app-server
ubuntu@ip-172-31-24-151:~$ lora-app-server
INFO[0000] starting LoRa App Server                      docs="https://docs.loraserver.io/" version=0.9.1
INFO[0000] connecting to postgresql
INFO[0000] setup redis connection pool
INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://localhost:1883"
INFO[0000] connecting to network-server api              ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
INFO[0000] applying database migrations
INFO[0000] handler/mqtt: connected to mqtt broker
INFO[0000] handler/mqtt: subscribling to tx topic        topic="application/+/node/+/tx"
FATA[0000] applying migrations failed: pq: relation "node" does not exist handling 0002_join_accept_params.sql
ubuntu@ip-172-31-24-151:~$

brocaar, 2017-05-19:
Are you using two separate databases, one for LoRa Server, one for LoRa App Server?

duy112, 2017-05-19:
no. i using 1 database for loraseerver and lora app server
but i tried using 2 databse and 1 had log : 

ubuntu@ip-172-31-24-151:~$ lora-app-server
INFO[0000] starting LoRa App Server                      docs="https://docs.loraserver.io/" version=0.9.1
INFO[0000] connecting to postgresql
INFO[0000] setup redis connection pool
INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://localhost:1883"
INFO[0000] connecting to network-server api              ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
INFO[0000] applying database migrations
INFO[0000] handler/mqtt: connected to mqtt broker
INFO[0000] handler/mqtt: subscribling to tx topic        topic="application/+/node/+/tx"
INFO[0000] migrations applied                            count=16
INFO[0000] gwmigrate: migrating gateway data from LoRa Server
INFO[0000] starting application-server api               bind="0.0.0.0:8001" ca-cert= tls-cert= tls-key=
FATA[0000] --jwt-secret must be set
ubuntu@ip-172-31-24-151:~$

duy112, 2017-05-19:
```
my config  : 
# postgresql dsn (e.g.: postgres://user:password@hostname/database?sslmode=disable) (default: "postgres://localhost/loraserver?sslmode=disable")
POSTGRES_DSN=postgres://loraserver1:*****@localhost/loraserver1?sslmode=disable

# automatically apply database migrations
DB_AUTOMIGRATE=true

# redis url (e.g. redis://user:password@hostname/0) (default: "redis://localhost:6379")
REDIS_URL=redis://localhost:6379

# mqtt server (e.g. scheme://host:port where scheme is tcp, ssl or ws) (default: "tcp://localhost:1883")
MQTT_SERVER=tcp://localhost:1883

# mqtt server username (optional)
MQTT_USERNAME=

# mqtt server password (optional)
MQTT_PASSWORD=

# ca certificate used by the api server (optional)
CA_CERT=

# tls certificate used by the api server (optional)
TLS_CERT=

# tls key used by the api server (optional)
TLS_KEY=

# ip:port to bind the api server (default: "0.0.0.0:8001")
BIND=0.0.0.0:8001

# ip:port to bind the (user facing) http server to (web-interface and REST / gRPC api) (default: "0.0.0.0:8080")
HTTP_BIND=0.0.0.0:8080

# http server TLS certificate
HTTP_TLS_CERT=/etc/lora-app-server/certs/http.pem
# http server TLS key
HTTP_TLS_KEY=/etc/lora-app-server/certs/http-key.pem
# JWT secret used for api authentication / authorization
# You could generate this by executing 'openssl rand -base64 32' for example
JWT_SECRET=

# hostname:port of the network-server api server (default: "127.0.0.1:8000")
NS_SERVER=127.0.0.1:8000

# ca certificate used by the network-server client (optional)
NS_CA_CERT=

# tls certificate used by the network-server client (optional)
NS_TLS_CERT=

# tls key used by the network-server client (optional)
NS_TLS_KEY=

# the number of iterations used to generate the password hash
PW_HASH_ITERATIONS=100000

# debug=5, info=4, warning=3, error=2, fatal=1, panic=0
LOG_LEVEL=4

```

brocaar, 2017-05-19:
Then that is the issue, you must use two separate database. In this log, please see:

```
FATA[0000] --jwt-secret must be set
```

You must set that value.

duy112, 2017-05-19:
i very thank for your help!
i don't know how to create jwt-secret

i have a vps amaon and i login with *.ppk
please help me.
thank you

brocaar, 2017-05-19:
See the configuration file you just posted! It contains an example command:

```
# JWT secret used for api authentication / authorization
# You could generate this by executing 'openssl rand -base64 32' for example
JWT_SECRET=
```

closed 2017-05-19 by brocaar

duy112, 2017-05-19:
i start success
very thanks for your helps