brocaar/chirpstack-application-server #90: Considering adding Kafka producer

manland, 2017-05-18:
Hi,

Thanks for sharing this tool. We will use it intensively at [itk](http://www.itk.fr/) .

As our processing jobs communicate over kafka, using the application server requires us to have a small application pulling data from mqtt and pushing it back it to our kafka.
To avoid an unnecessary bridge, could we imagine adding a kafka producer at the lora-app-server output level ?

This producer will be optional and triggered by configuration not to break the current use of loraserver. Would you accept a pull-request implementing this?

We think that app-server output consumers would be very intrested in this as kafka enables them to have persistence over their messages. It could be very helpful when a consumer needs to updated or restarted, to ensure no messages will be lost.

Furthermore, as you may know, kafka is one of the main tools used in the iot world, that's why this new feature would open loraserver to a wider range of iot actors.

This could be the starting point of using kafka among this stack. We could even imagine to replace all mqtt channels by kafka ones. Indeed, updating loraserver for example forces us to lose packages. I know we can configure mqtt qos like it has been requested in [lora-gateway-bridge/#24](https://github.com/brocaar/lora-gateway-bridge/issues/24) but we think it would be a better idea to switch to a distributed queue with persistence.

On the other hand, the deployement of production ready loraserver, will be painful. But we think it's worth it !

By the way, we have open-sourced [lorhammer](http://lorhammer.itk.fr/) : a tool that stresses your loraserver, and the results are pretty amazing comparing to others network-server

[image]

brocaar, 2017-05-19:
First of all, thanks for sharing the Lorhammer tool, that is absolutely great! I'll try that out soon 😄 

Regarding your question, yes I could see an usecase where LoRa App Server would be able to integrate directly with other services. I'm currently in the process of adding a http integration, so that it would be possible to integrate with REST apis (for example MyDevices Cayenne https://mydevices.com). Before starting any work, please wait with any work until that is in place as it will be the foundation for other integrations.

I would see integration support mainly for stateless integrations (e.g. HTTP based), as this will be a configuration per application, and there could be many applications out there so ideally the server would not maintain a persistent connection for each integration. Would that work with Kafka (I must say that I never used it myself)?

Putting Kafka inbetween the LoRa Server components might be a bit overkill, given that Mosquitto is really lightweight. Note that MQTT is a main tool used in the IoT world too ;-) Also, you are still able to setup multiple LoRa Server instances to distribute the load and availability! LoRa Server uses distributed locks using Redis, no duplicated work is done. When shutting down one instance, it will stop listening for new gateway packets and will wait until the last queue item has been processed, see also https://github.com/brocaar/loraserver/blob/master/internal/uplink/uplink.go#L39.

manland, 2017-05-22:
Thanks for lorhammer we hope it will be usefull ;)

Ok, we will wait until the http integration is ready. But, kafka is different from http, we can see it more as playing the role of mqtt but with a persistence that enables performance.

sh4nnongoh, 2017-08-01:
Wow thanks for the Lorhammer! 

Yep it would be great to have the App Server act as a Kafka producer as well. 
There is no need to put kafka in between the components though.

fifthaxe, 2018-02-12:
Hi everyone,

HTTP integration has been done, any update / plan regarding this feature request ?

Thanks a lot

brocaar, 2018-02-12:
I don't have any experience (yet) with Kafka. Does a Kafka producer have a persistent connection with the broker, or is it stateless like HTTP?

In case of the first, I would see it as a backend that can be configured through the (new) configuration files. If it is something stateless, then it could potentially be configured per application. E.g. you don't want to end up with the case where you have thousands of applications, each having their own persistent TCP connection to their integrations. 

If you look at the new configuration, then this is already modeled for these kind of things :-)

```toml
[application_server.integration]
backend="kafka"

  # Kafka integration ...
  [application_server.integration.kafka]
  # settings here....

  # MQTT integration configuration used for publishing (data) events
  # and scheduling downlink application payloads.
  # Next to this integration which is always available, the user is able to
  # configure additional per-application integrations.
  [application_server.integration.mqtt]
  # MQTT server (e.g. scheme://host:port where scheme is tcp, ssl or ws)
  server="tcp://localhost:1883"

  # Connect with the given username (optional)
  username=""

  # Connect with the given password (optional)
  password=""

  # CA certificate file (optional)
  #
  # Use this when setting up a secure connection (when server uses ssl://...)
  # but the certificate used by the server is not trusted by any CA certificate
  # on the server (e.g. when self generated).
  ca_cert=""

  # TLS certificate file (optional)
  tls_cert=""

  # TLS key file (optional)
  tls_key=""
```

https://docs.loraserver.io/lora-app-server/install/config/

Configuring Kafka as an integration would then look for example like:

```toml
  # MQTT integration configuration used for publishing (data) events
  # and scheduling downlink application payloads.
  # Next to this integration which is always available, the user is able to
  # configure additional per-application integrations.
  [application_server.integration.mqtt]
  # MQTT server (e.g. scheme://host:port where scheme is tcp, ssl or ws)
  server="tcp://localhost:1883"

  # Connect with the given username (optional)
  username=""

  # Connect with the given password (optional)
  password=""

  # CA certificate file (optional)
  #
  # Use this when setting up a secure connection (when server uses ssl://...)
  # but the certificate used by the server is not trusted by any CA certificate
  # on the server (e.g. when self generated).
  ca_cert=""

  # TLS certificate file (optional)
  tls_cert=""

  # TLS key file (optional)
  tls_key=""
```

fifthaxe, 2018-02-12:
I see Kafka really usefull in two parts :
- as a legacy outgoing integration (because it's stateless like HTTP) configured per application.
=> would open lora-app-server to Big Data stream ingestion (Kafka is also used massively in Big Data world).
The second part is also an important feature to me :
- as an alternative way for lora-app-server of dealing with data integration (think to an integration handler).
=> Administrator would be able select an integration handler, choosing kafka will send whole traffic to kafka, publishing to topics corresponding to integration configured per application.
This way, AS would deleguate data integration to kafka, providing massive scale benefits.

Let me know what do you think about that (may raise a new issue for part#2)

manland, 2018-02-13:
Hi,

Kafka is like mqtt (you can't open 10000 connections if you have 10000 applications). 

It has topics like mqtt, the main difference is you can't listen a topic with wildcard (devices/+/...). And kafka don't support lot of topic in parallel.

If we start en implementation, I think the best option is to do a topic per root route of mqtt (devices/, applications/, ...). What do you think about that ? The message contains all informations, the application next to loraserver will need (devEUI, applpicationID...)

IMHO if we add kafka backend, its a big plus for loraserver and the big data industry ;) Kafka manage back-pressure, permit horizontal scaling of readers and writers...

If you are ok to accept a pull-request, I will discuss with my team when we can do it.

brocaar, 2018-02-13:
> Kafka is like mqtt (you can't open 10000 connections if you have 10000 applications).

In that case I suggest to make this a backend that can be configured though the config files, so either you globally use MQTT or Kafka (or an other backend I'm currently working on 😉 ).

Ideally you would have a single connection from LoRa App Server (per instance) to Kafka. E.g. when you have 1000 users with each 10 applications, when each of them select the Kafka integration for their applications, each LoRa App Server **instance** has 1000 x 10 connections. So when you have 10 LoRa App Server instances, you have 1000 x 10 x 10 open TCP connections. I don't think this scales well ;-)

A pull-request for such backend would be appreciated @manland !

fifthaxe, 2018-02-15:
What about NATS (https://nats.io/) support as a backend ?

Hienphamthe, 2019-06-30:
Hi, 
Is this feature available or there have already been other workaround?

brocaar, 2019-09-16:
I'm closing this issue as there hasn't been any progress. If somebody is (still) interested in adding this feature, then I would welcome a pull-request. Happy to review and merge it!

closed 2019-09-16 by brocaar