brocaar/chirpstack-application-server #93: Send Class C downlink messages through MQTT

gkov9, 2017-05-31:
Hello Orne,

This is really a "feature" comment and not an "issue" one. I would suggest to handle all payload functions to be handled "centrally". Today, Class A up and downlink is handled through MQTT, while Class C downlink is handled by LoRa App Server API. Don't you think it would be more useful to have a one channel payload handling (trough MQTT)?

brocaar, 2017-05-31:
Please note that both Class-A and Class-C can be scheduled through the gRPC / REST API and MQTT. Under the hood, the scheduling is handled centrally. These are just two different interfaces to schedule.

gkov9, 2017-06-04:
Hello,

Does this mean that if I publish a tx message to MQTT (and the end node is set to Class C) then the message is sent immediatelly through the gateway?

brocaar, 2017-06-19:
@gkov9, yes that is correct.

ehsanmqn, 2017-09-03:
Dear @brocaar 
I am wondering about the structure of tx messages. When I use the following format, tx message will show in the application server frame log section:
{
    "fcnt": 3,
    "fport": 1,
    "applicationID": "1",
    "applicationName": "environment-sensor",
    "nodeName": "node1",
    "devEUI": "0102030405060709",
    "data": "MjUsMTUsICAgICAgIDAsICAgICAgIDAsICAg"
}

but for the following structure data does not appear on the application server. It seems that data does not send at all. Please guide me and if is it possible give me a good reference about these topics.

ehsanmqn, 2017-09-03:
Dear @brocaar 
I am wondering about the structure of tx messages. When I use the following format, tx message will show in the application server frame log section:
{
    "fcnt": 3,
    "fport": 1,
    "applicationID": "1",
    "applicationName": "environment-sensor",
    "nodeName": "node1",
    "devEUI": "0102030405060709",
    "data": "MjUsMTUsICAgICAgIDAsICAgICAgIDAsICAg"
}

but for the following structure data does not appear on the application server. It seems that data does not send at all. Please guide me and if is it possible give me a good reference about these topics.
{
    "data": "MjUsMTUsICAgICAgIDAsICAgICAgIDAsICAg"
}

closed 2017-09-04 by brocaar