brocaar/chirpstack-application-server #94: make api; make; Then make failure

tq7896, 2017-06-04:
hello: 
      brocaar
i got a error message when compiling the source.
failure message:
api/application.pb.gw.go:37: cannot use "github.com/brocaar/lora-app-server/vendor/google.golang.org/grpc/codes".InvalidArgument (type "github.com/brocaar/lora-app-server/vendor/google.golang.org/grpc/codes".Code) as type "google.golang.org/grpc/codes".Code in argument to "google.golang.org/grpc/status".Errorf
source code: application.pb.gw.go: line 36-38
if err := marshaler.NewDecoder(req.Body).Decode(&protoReq); err != nil {
		return nil, metadata, status.Errorf(codes.InvalidArgument, "%v", err)
	}

is grpc too old？？

utopiaprince, 2017-06-06:
you grpc is new. and protoc-gen-grpc-gateway is old.

tq7896, 2017-06-09:
What should I do? Upgrade or downgrade? thanks

grpc: v3.0.0

the cmd of gateway:
go get github.com/grpc-ecosystem/grpc-gateway/protoc-gen-grpc-gateway

utopiaprince, 2017-06-12:
in api/gen.sh, you could see that:
```
# generate the gRPC code
protoc -I/usr/local/include -I. ${GOPATHLIST} --go_out=Mgoogle/api/annotations.proto=github.com/grpc-ecosystem/grpc-gateway/third_party/googleapis/google/api,plugins=grpc:. channelList.proto node.proto application.proto downlinkQueue.proto common.proto user.proto gateway.proto
```
and this param is point:
```
--go_out=Mgoogle/api/annotations.proto=github.com/grpc-ecosystem/grpc-gateway/third_party/googleapis/google/api,plugins=grpc:.
```
specified the "annotations.proto" file.
 
event you "go get github.com/grpc-ecosystem/grpc-gateway/protoc-gen-grpc-gateway", get package "grpc-gateway" in $GOPATH, the vendor folder under root of project still have another version of grpc-gateway.

so you can use the older grpc to match the old grpc-gateway under vendor folder. or delete the grpc-gateway under folder.

closed 2017-07-25 by brocaar