brocaar/chirpstack-application-server #97: lora-app-server.service: Main process exited, code=exited, status=1/FAILURE

lytkarinskiy, 2017-06-16:
Hello! 
I can start loraserver, but can't start lora-app-server.
I use 2 different DB, and I've generated JWT_SECRET.
Can you help me to diagnose the problem? =)

Thanks

Versions

>lora-app-server/unknown,now 0.10.0 amd64 [installed]
loraserver/unknown,now 0.18.0 amd64 [installed]

Here I attach log files both for lora-app-server and loraserver.
### lora-app-server
>Jun 16 14:36:26 loraserver01 lora-app-server[2463]: time="2017-06-16T14:36:26+03:00" level=info msg="starting LoRa App Server" docs="https://docs.loraserver.io/" version=0.10.0
Jun 16 14:36:26 loraserver01 lora-app-server[2463]: time="2017-06-16T14:36:26+03:00" level=info msg="connecting to postgresql"
Jun 16 14:36:26 loraserver01 lora-app-server[2463]: time="2017-06-16T14:36:26+03:00" level=info msg="setup redis connection pool"
Jun 16 14:36:26 loraserver01 lora-app-server[2463]: time="2017-06-16T14:36:26+03:00" level=info msg="handler/mqtt: connecting to mqtt broker" server="tcp://localhost:1883"
Jun 16 14:36:26 loraserver01 lora-app-server[2463]: time="2017-06-16T14:36:26+03:00" level=info msg="connecting to network-server api" ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
Jun 16 14:36:26 loraserver01 lora-app-server[2463]: time="2017-06-16T14:36:26+03:00" level=info msg="applying database migrations"
Jun 16 14:36:26 loraserver01 lora-app-server[2463]: time="2017-06-16T14:36:26+03:00" level=info msg="handler/mqtt: connected to mqtt broker"
Jun 16 14:36:26 loraserver01 lora-app-server[2463]: time="2017-06-16T14:36:26+03:00" level=info msg="handler/mqtt: subscribling to tx topic" topic="application/+/node/+/tx"
Jun 16 14:36:26 loraserver01 systemd[1]: lora-app-server.service: Main process exited, code=exited, status=1/FAILURE
Jun 16 14:36:26 loraserver01 systemd[1]: lora-app-server.service: Unit entered failed state.
Jun 16 14:36:26 loraserver01 systemd[1]: lora-app-server.service: Failed with result 'exit-code'.
Jun 16 14:36:26 loraserver01 systemd[1]: lora-app-server.service: Service hold-off time over, scheduling restart.
Jun 16 14:36:26 loraserver01 systemd[1]: Stopped LoRa App Server.
Jun 16 14:36:26 loraserver01 systemd[1]: lora-app-server.service: Start request repeated too quickly.
Jun 16 14:36:26 loraserver01 systemd[1]: Failed to start LoRa App Server.

### loraserver
> Jun 16 14:36:29 loraserver01 systemd[1]: Started LoRa Server.
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="starting LoRa Server" band="EU_863_870" docs="https://docs.loraserver.io/" net_id=010203 version=0.18.0
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="setup redis connection pool" url="redis://localhost:6379"
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="connecting to postgresql"
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="backend/gateway: connecting to mqtt broker" server="tcp://localhost:1883"
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="connecting to application-server" ca-cert= server="127.0.0.1:8001" tls-cert= tls-key=
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="no network-controller configured"
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="backend/gateway: connected to mqtt server"
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="backend/gateway: subscribing to rx topic" topic="gateway/+/rx"
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="backend/gateway: subscribing to stats topic" topic="gateway/+/stats"
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="migrated node-sessions to new format" errors=0 migrated=0
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="applying database migrations"
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8001 <nil>}"
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="migrations applied" count=0
Jun 16 14:36:29 loraserver01 loraserver[2479]: time="2017-06-16T14:36:29+03:00" level=info msg="starting api server" bind="0.0.0.0:8000" ca-cert= tls-cert= tls-key=
Jun 16 14:36:30 loraserver01 loraserver[2479]: time="2017-06-16T14:36:30+03:00" level=info msg="grpc: addrConn.resetTransport failed to create client transport: connection error: desc = "transport: dial tcp 127.0.0.1:8001: getsockopt: connection refused"; Reconnecting to {127.0.0.1:8001 <nil>}"

### config for lora-appserver
`# postgresql dsn (e.g.: postgres://user:password@hostname/database?sslmode=disable) (default: "postgres://localhost/loraserver?sslmode=disable")`
`POSTGRES_DSN=postgres://loraappserver:****@localhost/loraappserver?sslmode=disable`

`# automatically apply database migrations`
`DB_AUTOMIGRATE=true`

`# redis url (e.g. redis://user:password@hostname/0) (default: "redis://localhost:6379")`
`REDIS_URL=redis://localhost:6379`

`# mqtt server (e.g. scheme://host:port where scheme is tcp, ssl or ws) (default: "tcp://localhost:1883")`
`MQTT_SERVER=tcp://localhost:1883`

`# mqtt server username (optional)`
`MQTT_USERNAME=loraappserver`

`# mqtt server password (optional)`
`MQTT_PASSWORD=***`

`# ca certificate used by the api server (optional)`
`CA_CERT=`

`# tls certificate used by the api server (optional)
`TLS_CERT=`

`# tls key used by the api server (optional)`
`TLS_KEY=`

`# ip:port to bind the api server (default: "0.0.0.0:8001")`
`BIND=0.0.0.0:8001`

`# ip:port to bind the (user facing) http server to (web-interface and REST / gRPC api) (default: "0.0.0.0:8080")`
`HTTP_BIND=0.0.0.0:8080`

`# http server TLS certificate`
`HTTP_TLS_CERT=/etc/lora-app-server/certs/http.pem`

`# http server TLS key`
`HTTP_TLS_KEY=/etc/lora-app-server/certs/http-key.pem`

`# JWT secret used for api authentication / authorization`
`# You could generate this by executing 'openssl rand -base64 32' for example`
`JWT_SECRET=pzySgM31F8SIkiYk**********`

`# hostname:port of the network-server api server (default: "127.0.0.1:8000")`
`NS_SERVER=127.0.0.1:8000`

`# ca certificate used by the network-server client (optional)`
`NS_CA_CERT=`

`# tls certificate used by the network-server client (optional)`
`NS_TLS_CERT=`

`# tls key used by the network-server client (optional)`
`NS_TLS_KEY=`

`# the number of iterations used to generate the password hash`
`PW_HASH_ITERATIONS=100000`

`# debug=5, info=4, warning=3, error=2, fatal=1, panic=0`
`LOG_LEVEL=5`

### config loraserver - only part where i've done changes
`# network identifier (NetID, 3 bytes) encoded as HEX (e.g. 010203)`
`NET_ID=010203`

`# ism band configuration to use (options: AU_915_928, CN_470_510, EU_863_870, US_902_928)`
`BAND=EU_863_870`

`# ca certificate used by the api server (optional)`
`CA_CERT=`

`# tls certificate used by the api server (optional)`
`TLS_CERT=`

`# tls key used by the api server (optional)`
`TLS_KEY=`

`# ip:port to bind the api server (default: "0.0.0.0:8000")`
`BIND=0.0.0.0:8000`

`# redis url (e.g. redis://user:password@hostname:port/0) (default: "redis://localhost:6379")`
`REDIS_URL=redis://localhost:6379`

`# postgresql dsn (e.g.: postgres://user:password@hostname/database?sslmode=disable)`
`POSTGRES_DSN=postgres://loraserver:****@@localhost/loraserver?sslmode=disable`

`# automatically apply database migrations`
`DB_AUTOMIGRATE=true`

`# mqtt broker server used by the gateway backend (e.g. scheme://host:port where scheme is tcp, ssl or ws) (default: "tcp://localhost:1883")`
`GW_MQTT_SERVER=tcp://localhost:1883`

`# mqtt username used by the gateway backend (optional)`
`GW_MQTT_USERNAME=loraserver`

`# mqtt password used by the gateway backend (optional)`
`GW_MQTT_PASSWORD=****`

fvdnabee, 2017-06-16:
I had the same systemctl and journalctl output when failing to start lora-app-server, but unfortunately the output does not hint where the problem lies. I managed to get more useful information by starting lora-app-server manually (using the systemctl environment file) and checking its output, as follows (execute as root):

> root@ns:/etc# sh -ac ' . /etc/default/lora-app-server; /usr/bin/lora-app-server'
INFO[0000] starting LoRa App Server                      docs="https://docs.loraserver.io/" version=0.10.0
INFO[0000] connecting to postgresql                     
INFO[0000] setup redis connection pool                  
INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://localhost:1883"
INFO[0000] connecting to network-server api              ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
INFO[0000] handler/mqtt: connected to mqtt broker       
INFO[0000] handler/mqtt: subscribling to tx topic        topic="application/+/node/+/tx"
INFO[0000] applying database migrations                 
INFO[0000] migrations applied                            count=0
INFO[0000] gwmigrate: migrating gateway data from LoRa Server 
INFO[0000] starting application-server api               bind="0.0.0.0:8001" ca-cert= tls-cert= tls-key=
FATA[0000] --jwt-secret must be set

In my case JWT_SECRET was missing from my enviroment file and adding it fixed my issue. Judging from your environment file, the problem probably lies elsewhere. Initially I though the missing  'FATA' error output line in journalctl/systemctl is probably a bug in lora-app-server, though now I've found this [systemctl issue](https://github.com/systemd/systemd/issues/3277) so perhaps it is not.

lytkarinskiy, 2017-06-16:
fvdnabee, thank you! 

so here is my output from proposed procedure, seems some access failure, but I double checked all passwords and it should be fine. Any suggestions ?

>root@loraserver01:/home/loradmin# sh -ac ' . /etc/default/lora-app-server; /usr/bin/lora-app-server'
INFO[0000] starting LoRa App Server                      docs="https://docs.loraserver.io/" version=0.10.0
INFO[0000] connecting to postgresql
INFO[0000] setup redis connection pool
INFO[0000] handler/mqtt: connecting to mqtt broker       server="tcp://localhost:1883"
INFO[0000] connecting to network-server api              ca-cert= server="127.0.0.1:8000" tls-cert= tls-key=
INFO[0000] handler/mqtt: connected to mqtt broker
INFO[0000] handler/mqtt: subscribling to tx topic        topic="application/+/node/+/tx"
INFO[0000] applying database migrations
FATA[0000] applying migrations failed: pq: permission denied for relation gorp_migrations

fvdnabee, 2017-06-16:
@lytkarinskiy a quick google shows that your issue might be related to this one https://github.com/brocaar/loraserver/issues/155

lytkarinskiy, 2017-06-16:
@fvdnabee thanks for the link!

It says I should check db-automigrate option , but it's enabled in my config

> automatically apply database migrations
DB_AUTOMIGRATE=true

Also I checked psql permissions 
>                                      List of databases
>     Name      |     Owner     | Encoding |   Collate   |    Ctype    |   Access privileges
>---------------+---------------+----------+-------------+-------------+-----------------------
> loraappserver | loraappserver | UTF8     | en_US.UTF-8 | en_US.UTF-8 |
> loraserver    | loraserver    | UTF8     | en_US.UTF-8 | en_US.UTF-8 |

(sorry for formatting)

And password seems OK.

>POSTGRES_DSN=postgres://loraappserver:****@localhost/loraappserver?sslmode=disable

lytkarinskiy, 2017-06-19:
So I reinstalled everything and it just works... sorry, I missed somewhere

closed 2017-06-19 by lytkarinskiy