brocaar/chirpstack-application-server #98: Consider move json struct models out of internals package

muka, 2017-06-19:
Hi
just a minor feedback for Go development. Using a go mqtt client I look to get the messages from app server, but `github.com/brocaar/lora-app-server/internal/handler` cannot be reused as under `internal` so afaik the only option is to copy paste the structures in my own project

(Reference `https://github.com/brocaar/lora-app-server/blob/master/internal/handler/models.go`)

Thank you

pmiam999, 2017-10-25:
One fix may be to soft link **internal** folder to **lib** in order to avoid copy paste and permit external usage.

chopmann, 2019-10-10:
@brocaar this or a swagger definition for the http-integration would be awesome

brocaar, 2019-11-13:
This has been implemented as https://github.com/brocaar/chirpstack-api. In the latest master branch it is possible to use a `json` or `protobuf` integration marshaler, which allows you to use the https://github.com/brocaar/chirpstack-api/blob/master/protobuf/as/integration/integration.proto definition at the client side.

The current default `json_v3` marshaler will use the "current" version of the JSON structs. This will be dropped in the next major release in favor of the Protobuf based binary (`protobuf`) or JSON (`json`) format.

closed 2019-11-13 by brocaar